myriadrf/LimeSuite #269: Mini becomes unusable - SoapySDR.Device_getFrequency blocking

sw, 2019-07-18:
We are using a LimeSDR Mini 1.2 with a recent LimeSuite (54614da) via SoapySDR.

Our problem is that the device becomes unresponsive. The Python interpreter will lock up in `_SoapySDR.Device_getFrequency`.

Once this has happened, LimeUtil will still find the device but not show the name:
```
C:\Work\TAS\lime>LimeUtil.exe --find
  * [, media=USB, module=FT601, index=0]
```

After removing the LimeSDR and re-plugging it, the device is correctly identified and will work again:
```
C:\Work\TAS\lime>LimeUtil.exe --find
  * [LimeSDR Mini, media=USB 3, module=FT601, serial=1D42499EA18CD2, index=0]
```

Unfortunately `LimeQuickTest` doesn't print anything in either case, we will have to look into getting that working.

In order to pinpoint any problem on the PC side, we have added logging to LMS7002M.cpp as follows:
```
float_type LMS7002M::GetFrequencySX(bool tx)
{
    Log(LOG_INFO, "GetFrequencySX(tx=%d)", tx);
    Channel ch = this->GetActiveChannel(); //remember previously used channel
    float_type dMul;
    Log(LOG_INFO, "active channel %d", ch);
...
```
This showed us that the lockup always seems to occur in `GetActiveChannel`, because we see the first log line but not the second.

We will try to use a different USB hub and try to get `LimeQuickTest` working.

Is there anything else we can to do narrow down the source of the problem?

sw, 2019-07-18:
Issue with LimeQuickTest was due to a linker pragma -> #270

We will run the fixed test program once the lockup occurs again.

sw, 2019-07-18:
Slightly different symptoms this time: red LED is constantly on, device is recognized by LimeUtil but not by LimeQuickTest:
```
c:\Work\TAS\lime>LimeUtil.exe --find
  * [LimeSDR Mini, media=USB 3, module=FT601, serial=1D42499EA18CD2, index=0]

c:\Work\TAS\lime>LimeQuickTest.exe
[ TESTING STARTED ]
->Start time: Thu Jul 18 14:29:36 2019

Board not supported
Failed to connect

c:\Work\TAS\lime>
```
After re-plugging:
```
c:\Work\TAS\lime>LimeUtil.exe --find
  * [LimeSDR Mini, media=USB 3, module=FT601, serial=1D42499EA18CD2, index=0]

c:\Work\TAS\lime>LimeQuickTest.exe
[ TESTING STARTED ]
->Start time: Thu Jul 18 14:30:06 2019

->Device: LimeSDR Mini, media=USB 3, module=FT601, serial=1D42499EA18CD2, index=0
  Serial Number: 1D42499EA18CD2

[ Clock Network Test ]
->REF clock test
  Test results: 3860; 17057; 30254 - PASSED
->VCTCXO test
  Results : 6711077 (min); 6711237 (max) - PASSED
->Clock Network Test PASSED

[ FPGA EEPROM Test ]
->Read EEPROM
->Read data: 12 07 1A 12 07 1A 02
->FPGA EEPROM Test PASSED

[ LMS7002M Test ]
->Perform Registers Test
->External Reset line test
  Reg 0x20: Write value 0xFFFD, Read value 0xFFFD
  Reg 0x20: value after reset 0x0FFFF
->LMS7002M Test PASSED

[ RF Loopback Test ]
->Configure LMS
->Run Tests (TX_2 -> LNA_W):
  CH0 (SXR=1000.0MHz, SXT=1005.0MHz): Result:(-14.1 dBFS, 5.00 MHz) - PASSED
->Run Tests (TX_1 -> LNA_H):
  CH0 (SXR=2100.0MHz, SXT=2105.0MHz): Result:(-14.5 dBFS, 5.00 MHz) - PASSED
->RF Loopback Test PASSED

=> Board tests PASSED <=

Elapsed time: 6.79 seconds

c:\Work\TAS\lime>
```

IgnasJarusevicius, 2019-07-19:
Looks like communication problem. test probably gets garbage response when checking board id and concludes that board is not supported. Do you use the latest FTDI driver from their site? If not try installing it.

sw, 2019-07-26:
This problem keeps occurring. We do have the latest FTDI driver (1.3.0.2) and we have also swapped out the USB 3 hub.

When the problem occurs, the Python process hangs and cannot be killed in Task Manager.

Because of the hanging process, it seems that LimeQuickTest can't access the device in this state:
```
C:\Users\gruppeesw>c:\work\TAS\lime\LimeUtil.exe --find
  * [, media=USB, module=FT601, index=0]
  * [, media=USB, module=FT601, index=1]

C:\Users\gruppeesw>c:\work\TAS\lime\LimeQuickTest.exe
[ TESTING STARTED ]
->Start time: Fri Jul 26 08:22:32 2019

Warning: Multiple devices detected, connecting to the first device
Failed to list USB Devices
Failed to open device
Failed to open. Device is busy.
Error: Unable to connect
Failed to connect
```
Unplugging the board kills the Python process, and after re-plugging things will work again for some time.

sw, 2019-07-30:
The process seems to get blocked in the following code path:
`LMS7002M::GetFrequencySX` -> `LMS7002M::GetActiveChannel` -> `LMS7002M::Get_SPI_Reg_bits` -> `LMS7002M::SPI_read` -> `LMS7002M::SPI_read_batch` -> `LMS64CProtocol::ReadLMS7002MSPI` -> `LMS64CProtocol::TransferPacket`

I have added logging statements to `TransferPacket` and other methods, and now the problem no longer occurs, which may point to some race condition / timing issue.

sw, 2019-07-31:
Observed a case where `GetOverlappedResult` in `ConnectionFT601::Read` returned FALSE, causing our pytest job to abort non-gracefully.
https://github.com/myriadrf/LimeSuite/blob/64bb2b89963ac48706e7990d1f3b60bf65fd5d33/src/ConnectionFTDI/ConnectionFT601.cpp#L343-L347

In another case, while reading MAC, the return value of `FT_ReadPipe` was not `FT_IO_PENDING`.
https://github.com/myriadrf/LimeSuite/blob/64bb2b89963ac48706e7990d1f3b60bf65fd5d33/src/ConnectionFTDI/ConnectionFT601.cpp#L333-L338
Subsequently, `ConnectionFT601::WaitForSending` was called with `contextHandle` = -1, which causes an invalid access of `contextsToSend`.

sw, 2019-07-31:
Generally speaking, error handling is often poor or missing. For example, `Streamer::TransmitPacketsLoop` does not check if `ConnectionFT601::BeginDataSending` returns a valid handle, but blindly sets `bufferUsed[bi] = true;`

sw, 2019-08-05:
Some more failure observations concerning `ConnectionFT601::Read`:
* `WaitForSingleObject` returns `WAIT_TIMEOUT`; we modified the source to call `GetLastError` which indicates `ERROR_IO_INCOMPLETE` (0x3E4)
* `WaitForSingleObject` returns `WAIT_OBJECT_0`; `GetLastError` indicates `ERROR_OPERATION_ABORTED` (0x3E3)

In both these cases, there are then more read errors indicated and the handle list seems to become corrupted:
* `WaitForSending` is being called with handle = -1
* `FinishDataSending` is being called with handle = -1
* `BeginDataSending` can't find an unused context

sw, 2019-08-09:
@IgnasJarusevicius can you or someone else from the Lime team please look into this?

We have bought 10 LimeSDR Mini and are planning to buy more, but the driver really needs to be robust.

I am no C++ expert, but FWIW it seems that using arrays and indices for {{USBTransferContext}} (in {{ConnectionFT601}} module) is a source of many potential bugs - maybe some other data structure would be better suited?

Thank you.

sw, 2019-08-23:
We have also seen LimeSuite locking up after sending a burst, at https://github.com/myriadrf/LimeSuite/blob/05d454e8848233e6c270596c0214d6a670c8bd75/src/protocols/Streamer.cpp#L547

I should add that we mostly see all these problems on a 32 bit Windows 7 machine that has several LimeSDR Minis attached, which we use concurrently.

Use of a single Mini on a Windows 10 machine has not lead to any problems, but we use it much less there, so that may just be statistics..

IgnasJarusevicius, 2019-08-29:
I have looked at ConnectionFTDI code and fixed some inconsistencies:
1. Non-negative handle check was performed only in WaitForReading and FinishDataReading but not  in WaitForSending and FinishDataSending
2. When transfer submit fails context.used was set to false on Linux but not on Windows;

The changes should improve error handling. However, in normal operation there should not be errors in the first place. The changes most likely will prevent crashes when transfer functions start producing errors, but will not make it work.

If you see something like:
` [, media=USB, module=FT601, index=0]` 
instead of:
` [LimeSDR Mini, media=USB 3, module=FT601, serial=1D42499EA18CD2, index=0]`
that means that LimeSuite is unable to even get device information from FTDI. It is likely that something in FTDI library or driver is hanging.

sw, 2019-09-05:
Thanks @IgnasJarusevicius, I've updated our setup to the latest master.

What we are now seeing is that `ConnectionFT601::Read` fails at `GetOverlappedResult`:
https://github.com/myriadrf/LimeSuite/blob/09445d3b5a67c70ca3cee2cbc226048ad60baa73/src/ConnectionFTDI/ConnectionFT601.cpp#L343-L347
`dwRet` is `WAIT_OBJECT_0`, `GetLastError()` returns 0x37 (`ERROR_DEV_NOT_EXIST`)

The code then calls `ReinitPipe`, but this does not seem to fix the problem, because subsequent calls to `ConnectionFT601::Read` fail at 
https://github.com/myriadrf/LimeSuite/blob/09445d3b5a67c70ca3cee2cbc226048ad60baa73/src/ConnectionFTDI/ConnectionFT601.cpp#L333-L338

After some retries of `Read` we then run into the error "ERROR BEGIN DATA SENDING 32" that you added recently:
https://github.com/myriadrf/LimeSuite/commit/7d86012fd97c00b2b83ba198a39b14f4ec72d185#diff-0c722ff58ed762aa13e907c941d2261bR597

sw, 2019-10-01:
Problem persists after upgrading the OS:
* before: Windows 7 32bit, Python 3.6 32 bit
* now: Windows 10 64bit, Python 3.7 32 bit

Mostly we are now seeing error 0x3E4 (ERROR_IO_INCOMPLETE) for `GetOverlappedResult` in `ConnectionFT601::Read`.

Increasing the default timeout from 100ms to 2s did not help

sw, 2019-10-04:
So the problem is in the handling of the return value of `ConnectionFT601::Write` by `LMS64CProtocol::TransferPacket`. It checks if the return value is different from zero, but -1 is used to indicate an error. `TransferPacket` then happily calls `ConnectionFT601::Read`, which blocks.

@IgnasJarusevicius Please consider the attached patch for inclusion.
Aside from fixing the `TransferPacket` function, I have also removed the call to `WaitForSingleObject` in the Read and Write functions. It seems to me that having two concurrent timeouts (the pipe timeout in the FTDI driver and the one enforced by WaitForSingleObject) is a recipe for confusion.
I have also cleaned up some redundant variables.
[limesuite-269.patch.txt](https://github.com/myriadrf/LimeSuite/files/3690935/limesuite-269.patch.txt)

We have not seen any lockups with the patched version but we do have some erroneous transmissions (possibly due to the host not being fast enough to serve multiple SDRs concurrently). I'll investigate further.

sw, 2019-10-10:
> FT_WritePipe() and FT_ReadPipe() should return immediately when asynchronous transfers are used (overlapped is not NULL) and then WaitForSingleObject is needed to wait for completion.

It's not really asynchronous if you're going to block anyway. That's why I thought I would simplify the code, but if that has caused problems maybe best leave it.

The read after failed write in `TransferPacket` is definitively a problem, please fix it.

The process still blocks sometimes in `SoapyLMS7::deactivateStream` -> `StreamChannel::Stop` -> `Streamer::UpdateThreads` -> `txThread.join();`
I haven't been able to dig into it further. I suspect some error case is not properly handled in `TransmitPacketsLoop`

I will try the beta driver.

sw, 2019-12-18:
Our LimeSDR Minis work better now. There are still problems sometimes, but this may be due to problems in the Windows driver for the USB hub or the FTDI chip. The reason I believe that is because in some error cases, the device is still shown in Device Manager even after unplugging it. A reboot or unplugging of the hub is then required.

I'm closing this for now. Thanks for your help.

closed 2019-12-18 by sw