myriadrf/LimeSuite #273: undefined reference to `__atomic_load_8'

pespin, 2019-08-05:
We recently enabled building LimeSuite latest release in Osmocom's Raspbian10 platform (armv7l), so we can have nightly packages of osmo-trx-lms for that platform.

It seems, however, that linking fails on that distro+architecture:
https://build.opensuse.org/package/live_build_log/network:osmocom:nightly/limesuite/Raspbian_10/armv7l

```
[  229s] [ 40%] Linking CXX executable ../../bin/basicRX
[  229s] cd /usr/src/packages/BUILD/obj-arm-linux-gnueabihf/src/examples && /usr/bin/cmake -E cmake_link_script CMakeFiles/basicRX.dir/link.txt --verbose=1
[  229s] /usr/bin/c++  -g -O2 -fdebug-prefix-map=/usr/src/packages/BUILD=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fvisibility-inlines-hidden -pthread -O3 -DNDEBUG  -Wl,-z,relro -rdynamic CMakeFiles/basicRX.dir/basicRX.cpp.o  -o ../../bin/basicRX -Wl,-rpath,/usr/src/packages/BUILD/obj-arm-linux-gnueabihf/src ../libLimeSuite.so.19.04.0 -pthread -lusb-1.0 
[  230s] [ 41%] Linking CXX executable ../../bin/singleRX
[  230s] cd /usr/src/packages/BUILD/obj-arm-linux-gnueabihf/src/examples && /usr/bin/cmake -E cmake_link_script CMakeFiles/singleRX.dir/link.txt --verbose=1
[  230s] /usr/bin/ld: ../libLimeSuite.so.19.04.0: undefined reference to `__atomic_load_8'
[  230s] /usr/bin/ld: ../libLimeSuite.so.19.04.0: undefined reference to `__atomic_store_8'
[  230s] collect2: error: ld returned 1 exit status
[  230s] make[3]: *** [src/examples/CMakeFiles/basicRX.dir/build.make:89: bin/basicRX] Error 1
[  230s] make[3]: Leaving directory '/usr/src/packages/BUILD/obj-arm-linux-gnueabihf'
[  230s] /usr/bin/c++  -g -O2 -fdebug-prefix-map=/usr/src/packages/BUILD=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fvisibility-inlines-hidden -pthread -O3 -DNDEBUG  -Wl,-z,relro -rdynamic CMakeFiles/singleRX.dir/singleRX.cpp.o  -o ../../bin/singleRX -Wl,-rpath,/usr/src/packages/BUILD/obj-arm-linux-gnueabihf/src ../libLimeSuite.so.19.04.0 -pthread -lusb-1.0 
[  230s] make[2]: *** [CMakeFiles/Makefile2:407: src/examples/CMakeFiles/basicRX.dir/all] Error 2
[  230s] make[2]: *** Waiting for unfinished jobs....
[  230s] /usr/bin/ld: ../libLimeSuite.so.19.04.0: undefined reference to `__atomic_load_8'
[  230s] /usr/bin/ld: ../libLimeSuite.so.19.04.0: undefined reference to `__atomic_store_8'
[  230s] collect2: error: ld returned 1 exit status
[  230s] make[3]: *** [src/examples/CMakeFiles/singleRX.dir/build.make:89: bin/singleRX] Error 1
[  230s] make[3]: Leaving directory '/usr/src/packages/BUILD/obj-arm-linux-gnueabihf'
[  230s] make[2]: *** [CMakeFiles/Makefile2:370: src/examples/CMakeFiles/singleRX.dir/all] Error 2
```

So reading a bit on what could be the issue, it seems LimeSuite's CMake should have some checks added in order to see if libatomic needs to be linked in. Some references:
https://github.com/PurpleI2P/i2pd/issues/897
https://github.com/PurpleI2P/i2pd/issues/285
https://reviews.llvm.org/D10453

laf0rge, 2020-07-30:
It's almost one year later and LimeSuite dpkgs still fails to build on raspbian 10: https://build.opensuse.org/package/live_build_log/network:osmocom:nightly/limesuite/Raspbian_10/armv7l  - would be great if somebody could pick this up, particularly as the LimeNET-micro contains a Raspi compute module.

gracid, 2020-08-04:
Should be fixed with 39b3629b06c9aac064983b6975ab9a7c7e27b98d commit.

laf0rge, 2020-08-05:
20.07.1 still has the problem:
<pre>
CMakeFiles/LimeSuiteGUI.dir/limeRFE/limeRFE_cmd.cpp.o CMakeFiles/LimeSuiteGUI.dir/limeRFE/RFE_Device.cpp.o
-o ../bin/LimeSuiteGUI -Wl,-rpath,/usr/src/packages/BUILD/obj-arm-linux-gnueabihf/src:
libLimeSuite.so.20.07.0 oglGraph/liboglGraph.a -L/usr/lib/arm-linux-gnueabihf -pthread
-lwx_gtk2u_richtext-3.0 -lwx_baseu_xml-3.0 -lwx_gtk2u_html-3.0 -lwx_gtk2u_xrc-3.0 -lwx_gtk2u_adv-3.0
-lwx_gtk2u_aui-3.0 -lwx_gtk2u_core-3.0 -lwx_baseu-3.0 -lwx_gtk2u_gl-3.0 -lusb-1.0 -lGL
-lwx_gtk2u_richtext-3.0 -lwx_baseu_xml-3.0 -lwx_gtk2u_html-3.0 -lwx_gtk2u_xrc-3.0 -lwx_gtk2u_adv-3.0
-lwx_gtk2u_aui-3.0 -lwx_gtk2u_core-3.0 -lwx_baseu-3.0 -lwx_gtk2u_gl-3.0
[  806s] /usr/bin/ld: libLimeSuite.so.20.07.0: undefined reference to `__atomic_store_8'
[  806s] /usr/bin/ld: libLimeSuite.so.20.07.0: undefined reference to `__atomic_load_8'
[  806s] collect2: error: ld returned 1 exit status
[  806s] make[3]: *** [src/CMakeFiles/LimeSuiteGUI.dir/build.make:1005: bin/LimeSuiteGUI] Error 1
[  806s] make[3]: Leaving directory '/usr/src/packages/BUILD/obj-arm-linux-gnueabihf'
[  806s] make[2]: *** [CMakeFiles/Makefile2:145: src/CMakeFiles/LimeSuiteGUI.dir/all] Error 2
[  806s] make[2]: Leaving directory '/usr/src/packages/BUILD/obj-arm-linux-gnueabihf'
[  806s] make[1]: *** [Makefile:133: all] Error 2
[  806s] make[1]: Leaving directory '/usr/src/packages/BUILD/obj-arm-linux-gnueabihf'
[  806s] dh_auto_build: cd obj-arm-linux-gnueabihf && make -j2 returned exit code 2
[  806s] make: *** [debian/rules:28: binary] Error 2
[  806s] dpkg-buildpackage: error: fakeroot debian/rules binary subprocess returned exit status 2
[  806s] ### VM INTERACTION START ###
[  809s] [  781.138938] sysrq: SysRq : Power Off
[  809s] [  781.141870] reboot: Power down
[  809s] ### VM INTERACTION END ###
[  809s]
[  809s] armbuild02 failed "build limesuite_20.07.0-1.dsc" at Wed Aug  5 18:25:53 UTC 2020.
[  809s]
</pre>

But I just saw that your patch was exactly one commit after the 20.07.1 release.  At Osmocom we generally only build tagged releases of upstream dependencies like LimeSuite so far. Will have to see if we can add that one patch to the Debian package  build without too much impact to our build scripts.  Thanks anyway!

gracid, 2020-08-06:
Hello, just created the 20.07.2 tag which includes the Raspbian build problem fix. No need to patch the Debian package or change your current build scripts.

laf0rge, 2020-08-09:
I can confirm thsi fixed the Raspbian builds. Thanks!

closed 2020-08-12 by gracid