myriadrf/LimeSuite #276: Better documentation for LMS_SetLOFrequency multi channel limitations

pespin, 2019-08-21:
I'm trying to set up a 2 TRX BTS using osmo-trx and LimeSDR-USB using channels A and B. However, setting Rx frequency for channel B after having set it for channel A (different one) seems to fail:

```
Wed Aug 21 12:07:48 2019 DTRXCTRL <0001> Transceiver.cpp:773 [tid=140407242458880][chan=0] command is 'RXTUNE 1781800'
Wed Aug 21 12:07:48 2019 DDEV <0002> LMSDevice.cpp:797 [tid=140407242458880][chan=0] Setting Rx Freq to 1.7818e+09 Hz
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] INT 112, FRAC 2730, DIV_LOCH 1, EN_DIV2_DIVPROG 1
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] VCO 7127.20 MHz, RefClk 30.72 MHz
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] ICT_VCO: 192
Wed Aug 21 12:07:48 2019 DTRXCTRL <0001> Transceiver.cpp:773 [tid=140407234066176][chan=1] command is 'RXTUNE 1782200'
Wed Aug 21 12:07:48 2019 DDEV <0002> LMSDevice.cpp:797 [tid=140407234066176][chan=1] Setting Rx Freq to 1.7822e+09 Hz
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] TuneVCO(SXR) - VCO too low
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] VCOL : csw=0 tune fail
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] ICT_VCO: 192
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] TuneVCO(SXR) - VCO too high
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] VCOM : csw=0 tune fail
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] ICT_VCO: 192
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] INT 112, FRAC 16384, DIV_LOCH 1, EN_DIV2_DIVPROG 1
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] VCO 7128.00 MHz, RefClk 30.72 MHz
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] ICT_VCO: 192
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] TuneVCO(SXR) - VCO too low
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] VCOH : csw=0 tune fail
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] ICT_VCO: 224
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] TuneVCO(SXR) - VCO too low
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] VCOL : csw=0 tune fail
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] TuneVCO(SXR) - VCO too low
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] VCOL : csw=0 tune fail
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] ICT_VCO: 224
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] ICT_VCO: 224
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] TuneVCO(SXR) - VCO too low
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] VCOM : csw=0 tune fail
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] TuneVCO(SXR) - VCO too low
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] VCOM : csw=0 tune fail
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] ICT_VCO: 224
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] ICT_VCO: 224
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] TuneVCO(SXR) - VCO too high
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] VCOH : csw=0 tune fail
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] ICT_VCO: 255
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] csw=64      cmphl=0
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] csw=96      cmphl=0
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] csw=64      cmphl=0
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] csw=112     cmphl=0
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] csw=96      cmphl=0
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] csw=120     cmphl=0
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] csw=112     cmphl=0
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] csw=124     cmphl=0
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] csw=120     cmphl=0
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] csw=126     cmphl=0
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] csw=124     cmphl=0
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] csw=126     cmphl=0
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] csw=127     cmphl=0
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] Failed to lock
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] csw=127     cmphl=0
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] Failed to lock
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] csw=192     cmphl=3
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] csw=192     cmphl=3
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] csw=160     cmphl=3
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] csw=160     cmphl=3
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] csw=144     cmphl=2
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] csw=144     cmphl=2
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] csw=152     cmphl=2
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] csw=152     cmphl=2
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] csw=156     cmphl=2
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] csw=156     cmphl=2
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] csw=158     cmphl=3
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] csw=158     cmphl=3
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] csw=157     cmphl=3
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] csw=157     cmphl=3
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] Failed to lock
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] cmphl=2
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] VCOL : csw=139 tune ok
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] Failed to lock
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407242458880] ICT_VCO: 255
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] cmphl=0
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] TuneVCO(SXR) - failed to lock (cmphl!=2)
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] VCOH : csw=0 tune fail
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] Selected: VCOH
Wed Aug 21 12:07:48 2019 DLMS <0003> LMSDevice.cpp:94 [tid=140407234066176] SetFrequencySXR(1782 MHz) - cannot deliver frequency
Wed Aug 21 12:07:48 2019 DDEV <0002> LMSDevice.cpp:800 [tid=140407234066176][chan=1] Error setting Rx Freq to 1.7822e+09 Hz
Wed Aug 21 12:07:48 2019 DTRXCTRL <0001> Transceiver.cpp:852 [tid=140407234066176] RX failed to tune
```

Documentation for LMS_SetLOFrequency mentions that there's limitations when setting up frequency for both channels A and B, but it doesn't explictly state which kind of limitations:
```
 * @note channels A and B in LMS7 chip share the same clock so ability to set
 * different frequencies for channels A and B is very limited. This function
 * will attempt to achieve different requested frequencies using NCO when
 * possible, however often changing frequency for one (A or B) channel will
 * result in frequency being changed for both (A and B) channels.
```

Could you please document it in the API or at least insert some link to some place better explaining the limitations between both frequencies? 

I'd like to see if some ARFCN pair can be used, or whether I must implement multi-arfcn feature from osmo-trx used in Ettus B200 for osmo-trx-lms too in order to be able to run 2 TRX with LimeSDR devices (see https://ftp.osmocom.org/docs/latest/osmotrx-usermanual.pdf section 13.2  "Multi-ARFCN mode")

IgnasJarusevicius, 2019-08-29:
Well as it says, channels A and B uses the same PLL clock so different frequencies can only be achieved by shifting digitally (using NCO). Digital shift is limited by RF bandwidth that is being sampled.

laf0rge, 2019-08-29:
In other words, channels A and B are really only useful for MIMO and not for anything else, such as operating independend receivers/transmitters on different frequencies.

IgnasJarusevicius, 2019-08-29:
Yes, channels A and B are mainly meant to be used for MIMO. The actual RF PLL frequency will always be the same for both channels and the only possibility to transmit/receive at different frequencies is by utilizing hardware NCO and interpolation/decimation.