myriadrf/LimeSuite #283: after "sudo make install", command"install.sh", command not found

jin-github102, 2019-10-29:
after installingss, i met this.  Do you know whats wrong??

dell@dell-XPS-15-9550:~/LimeSuite/udev-rules$ ls
64-limesuite.rules  install.sh
dell@dell-XPS-15-9550:~/LimeSuite/udev-rules$ sudo ./install.sh
sudo: ./install.sh: command not found
dell@dell-XPS-15-9550:~/LimeSuite/udev-rules$ sudo install.sh
sudo: install.sh: command not found
dell@dell-XPS-15-9550:~/LimeSuite/udev-rules$ ls -l
total 8
-rw-rw-r-- 1 dell dell 471 Oct 29 18:46 64-limesuite.rules
-rw-rw-r-- 1 dell dell  99 Oct 29 18:45 install.sh

btashton, 2019-10-30:
@IgnasJarusevicius  Looks like when this restructure branch got merged in commit c02f4358b6eb3a83f333e8951419f34b18667dde some scripts lost there execute bit.

@jin-github102 `chmod +x ./install.sh` then try running it.

IgnasJarusevicius, 2019-10-30:
I have not noticed that execute flag got lost in that commit  about 1.5 years ago as I always used 'sudo sh install.sh'. I am going to add/bring back the flag.

closed 2019-10-30 by IgnasJarusevicius

jin-github102, 2019-10-30:
It works after using "sudo sh install.sh. 
I didnt try the other way. I think it will work too. 
Thank you very mush! @IgnasJarusevicius @btashton

commit 6ba8451f (2019-10-30): add execute flag to udev-rules/install.sh (fix #283)