myriadrf/LimeSuite #284: Soapy method "writeRegisters" does not seem to be implemented

KarlL2, 2019-11-13:
SoapyLMS7 implements `writeRegister` but not `writeRegisters`:

https://github.com/myriadrf/LimeSuite/blob/23b6172f4a7a89ffdee45937825717baffdaba1a/SoapyLMS7/SoapyLMS7.h#L240
https://github.com/myriadrf/LimeSuite/blob/4ab24db5783ebeaf9568b07e48e7a3e9e222ec38/SoapyLMS7/Settings.cpp#L703

Is it not possible to write to several registers on the Lime with the Soapy API? Or am I missing something?

This is related to the post [Frequency hopping transmit](https://discourse.myriadrf.org/t/frequency-hopping-transmit-fixed/3622/14?u=karll) where I would like to write 2 registers in one call.

KarlL2, 2019-11-18:
Could someone confirm or refute my finding? Maybe @IgnasJarusevicius or @guruofquality ?

IgnasJarusevicius, 2019-11-18:
It is not implemented in SoapyLMS7.
I do not know about SoapySDR itself, maybe it has default implementation that just calls writeRegister() multiple times.

KarlL2, 2019-11-19:
Thanks for the confirmation @IgnasJarusevicius .
It seems SoapySDR doesn't do anything by default:

https://github.com/pothosware/SoapySDR/blob/67abec9fe7ba84b4d069951c6e7f8c45f11e8335/lib/Device.cpp#L693

Any reasons why `writeRegisters` is not implemented in SoapyLMS7? Is it because `SPI_write_batch` takes an array of addresses and an array of data, while Soapy's API is a single address and array of data?

https://github.com/myriadrf/LimeSuite/blob/a5b3a10fe4db258b267d686a91b845e4375dfdc8/src/lms7002m/LMS7002M.cpp#L1981

If that's the case then the implementation of `writeRegisters` could create the corresponding array of contiguous addresses and call `SPI_write_batch`.