myriadrf/LimeSuite #289: FX3 bulk transfer timeout

TehWan, 2019-12-04:
There seems to be an underlying problem with the FX3 communication which causes it to timeout during a bulk transfer no matter the delay (I've tried 1000 ms to no avail). libusb thinks data has been transferred and fills the buffer with contents of the memory (security issue), even though the status is set to timed out. I only encounter this issue on Linux, not on Windows, which leads me to think it might not be related to the firmware on the FX3. This is related to #287.

TehWan, 2019-12-04:
Initial findings by listening to the USB packets: when it fails, the URB_BULK in packet returns a status of `-ENOENT`, or -2. Somehow, libusb considers this a timeout, but that's a different issue.

TehWan, 2019-12-04:
Turns out libusb is fine, confirmed by Wireshark. Somehow the leaked memory and the timeout are returned by the device, even if a timeout did not really occur. This might be an issue with the firmware, but it definitely warrants further investigation.

TehWan, 2019-12-04:
On Windows, the driver is Cypress USB 3 (CYUSB3), and the issue is not seen. On Linux, the driver is libusb-1.0 (built from latest source), and the issue occurs. This is looking more and more like an issue with the driver.

TehWan, 2019-12-04:
This is definitely an issue with the way libusb handles the communications for the bulk transfer. I found [previous reports](https://sourceforge.net/p/libusb/mailman/message/31959666/) attesting to that. I'll try to get a USB sniffer that does not depend on the driver/OS and see what's really going on.

IgnasJarusevicius, 2019-12-05:
Hello,
have you tried it on different PC? I do not see this issue on any of my Intel machines that run either Ubuntu 16.04 or Ubuntu 18.04. I remember seeing similar issue with another machine that had different USB 3.1 controller. The issue was that it would receive garbage data every other connect. It was something like: it is OK when opening LimeSDR-USB for the 1st time, then it fails when opening 2nd time and fine again on 3rd...

TehWan, 2019-12-05:
Hi,
On the same PC, it works fine through Windows using the Cypress USB 3 driver, and it has the issue on Ubuntu 18.04 using the libusb driver. I could not get the LimeSDR-USB to work on Windows with the libusb/WinUSB drivers. Both go through the same physical connection. I've also tried different connections (different hub) and did not notice any change.

I just tried on my laptop and the issue does not occur. You are correct in saying the USB controller might be to blame. Nonetheless, the driver should be able to cope with it, as the Cypress driver does. I don't think libusb is to blame either, as they really are receiving this junk data. This is where a hardware USB debugger would help me to figure out if the issue is in the kernel driver or on the hardware side.

The issue is definitely not related to the LimeSDR project and this issue can be closed. I'll raise an issue with the culprit after I get hardware debugging in place.

ddcc, 2019-12-05:
Out of curiosity, which USB hardware protocol analyzer are you planning to use? I had a Total Phase Beagle at a previous job, but it's too expensive for personal use :(

TehWan, 2019-12-05:
I couldn't find any that would fit both my needs and my budget. I was planning on making my own by simply sniffing the data lines and using software to convert the signals to bytes and such.

OhSoGood, 2020-01-20:
@TehWan : Would you have a sample code to run to verify the issue? I have two AMD desktop on which I was running the limesdr usb fine on Windows for months and now I'm facing issue like yours on #287 on Linux.

closed 2020-01-31 by TehWan