myriadrf/LimeSuite #292: [ERROR] Tx Calibration: MCU error 5 on LimeSDR-Mini

Highspeedy, 2020-01-12:
Hello,

on starting a program which is using SoapySDR with a LimeSDR-Mini as the SDR allways a Tx Calibration error is coming.  

-------- Error Log
Input
Type: edi
Source: tcp://localhost:9100
Output
SoapySDR
Device:
master_clock_rate: 32768000
Sampling rate: 4.0960 MHz
Soapy:Creating the device with:
RC: Waiting for connection on port 2121
[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D4C296EC121C3'
[INFO] Reference clock 40.00 MHz
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[INFO] LMS7002M calibration values caching Disable
SoapySDR:Actual master clock rate: 32767.9995 kHz
SoapySDR:Actual TX rate: 4096.0000 ksps.
[INFO] Selected TX path: Band 2
SoapySDR:Actual frequency: 223936.000 kHz.
SoapySDR:Actual TX gain: 40.00
SoapySDR:Actual TX antenna: BAND2
Setting up timestamp decoder with 0 offset
Opening EDI :tcp://localhost:9100
EDI TCP connect to localhost:9100
**[ERROR] Tx Calibration: MCU error 5 (Loopback signal weak: not connected/insufficient gain?)**
[INFO] Rx calibration finished
-------- End Error Log

The calibration process works without an error in Lime  Suite Gui (Ver. 19.04.0-gabf82140). The Log of Lime Suite Gui as an attachement Please rename the fielextension to bz2. GitHub did not support bz2-extension.

[LIMEMiniSDR_LimeSuiteGui_Log.txt.zip](https://github.com/myriadrf/LimeSuite/files/4050886/LIMEMiniSDR_LimeSuiteGui_Log.txt.zip)

Any ideas ?

Thanks, Martin

-------------------
Input
  Type: edi
  Source: tcp://localhost:9100
Output
 SoapySDR
  Device: 
  master_clock_rate: 32768000
  Sampling rate: 4.0960 MHz
      Soapy:Creating the device with: 
      RC: Waiting for connection on port 2121
[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D4C296EC121C3'
[INFO] Reference clock 40.00 MHz
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[INFO] LMS7002M calibration values caching Disable
      SoapySDR:Actual master clock rate: 32767.9995 kHz
      SoapySDR:Actual TX rate: 4096.0000 ksps.
[INFO] Selected TX path: Band 2
      SoapySDR:Actual frequency: 223936.000 kHz.
      SoapySDR:Actual TX gain: 40.00
      SoapySDR:Actual TX antenna: BAND2
      Setting up timestamp decoder with 0 offset
      Opening EDI :tcp://localhost:9100
      EDI TCP connect to localhost:9100
**[ERROR] Tx Calibration: MCU error 5 (Loopback signal weak: not connected/insufficient gain?)**
[INFO] Rx calibration finished

TehWan, 2020-02-04:
I found that in order to avoid this issue, I need to set the gain very high, then activate the streaming, then lower it down to the level I need. This seems to be necessary for TX calibration.

arunmagesh, 2021-08-14:
I have the same issue and nothing fixed my issues. Any pointer peeps???