myriadrf/LimeSuite #295: Bad readSetting() in SoapyLMS crashing CubicSDR

guruofquality, 2020-01-18:
CubicSDR wants to read the values for all advertised settings from getSettingInfo(), this includes SAVE_CONFIG, LOAD_CONFIG, and OVERSAMPLING. The following implementation of readSetting() simply throws or crashes.

```
std::string SoapyLMS7::readSetting(const std::string &key) const
{
    return readSetting(SOAPY_SDR_TX, 0, key);
}
```

on master branch of limesuite

IgnasJarusevicius, 2020-01-22:
The advertised settings are write-only. In most cases they were added to access some advanced parameter or functions and implementations of read-back was not required/requested. Attempting to read them should raise exception indicating failure/error ("unknown setting key"). 
So the easiest solutions are:
1. Hide read failure by returning empty string instead of raising exception in SoapyLMS
2. Or improve error/failure handling in CubicSDR

It is also possible to try to implement read-back of those setting with meaningful values when possible, but as some of those setting has been there for years without need for read-back, it seems like it may be too much effort for little/no benefit.

IgnasJarusevicius, 2020-01-23:
I implemented read for the settings advertised by getSettingInfo(). For load/save config it should now return empty string and in most other cases - cached write value. Can you check if this solves CubicSDR problem?

guruofquality, 2020-01-26:
@IgnasJarusevicius Thanks that works

> So the easiest solutions are:
    Hide read failure by returning empty string instead of raising exception in SoapyLMS
    Or improve error/failure handling in CubicSDR

Honestly I would do both :-)

closed 2020-01-26 by guruofquality