myriadrf/LimeSuite #296: error loading shared libraries when running LimeUtil on Fedora 31

linuxguy123, 2020-01-24:
I built LimeSuite using the instructions on this page: http://wiki.myriadrf.org/Lime_Suite#Unix_makefiles

Specifically, these steps:
git clone https://github.com/myriadrf/LimeSuite.git
cd LimeSuite
git checkout stable
mkdir builddir && cd builddir
cmake ../
make -j4
sudo make install
sudo ldconfig

When I run LimeUtil, I get the following:
$ LimeUtil 
LimeUtil: error while loading shared libraries: libLimeSuite.so.19.04-1: cannot open shared object file: No such file or directory

yet:
# find / -name libLimeSuite*
/home/me/Development/LimeSuite/build/src/libLimeSuite.so.19.04.1
/home/me/Development/LimeSuite/build/src/libLimeSuite.so
/home/me/Development/LimeSuite/build/src/libLimeSuite.so.19.04.0
/home/me/Development/LimeSuite/build/src/libLimeSuite.so.19.04-1
/usr/local/lib/libLimeSuite.so.19.04.0
/usr/local/lib/libLimeSuite.so.19.04-1
/usr/local/lib/libLimeSuite.so
/usr/local/lib/libLimeSuite.so.19.04.1

I tried moving them to /usr/local/lib64 and that made no difference.   Where are these library files supposed to be located ?

linuxguy123, 2020-01-24:
Could someone with a working LimeUtil command please run #find / -name libLimeSuite* on their computer and report the results back here ?

IgnasJarusevicius, 2020-01-29:
On Ubuntu the library is in "/usr/local/lib/" which is default cmake install directory and it works. 
I don't use Fedora so I don't know what directories it checks when searching for libraries.
You should be able to change cmake install path via DCMAKE_INSTALL_PREFIX option.

guruofquality, 2020-01-29:
When doing source builds on red hat systems, I usually set LD_LIBRARY_PATH in my bash profile to have /usr/local/lib and /usr/local/lib64

IgnasJarusevicius, 2020-02-24:
Closing as the issue seems to be problem with system library paths and not LimeSuite itself. Solution can probably be found by searching for information about library paths on Fedora. 
Also, we currently only test/support Ubuntu (16.04, 18.04).

closed 2020-02-24 by IgnasJarusevicius