myriadrf/LimeSuite #297: v20.01.0 fails to build with gcc and "-Werror=return-type"

mnhauke, 2020-01-28:
openSUSE Tumbleweed builds (gcc 9.2.1+) on [OBS](https://build.opensuse.org) are now forced to use "-Werror=return-type" which breaks compilation of the latest LimeSuite release:

```
[   35s] [ 26%] Building CXX object src/CMakeFiles/LimeSuite.dir/limeRFE/limeRFE_api.cpp.o
[   35s] cd /home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/build/src && /usr/bin/c++  -DLIME_DLL -DLIME_DLL_EXPORTS -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D__WXGTK__ -DwxDEBUG_LEVEL=0 -I/home/a
build/rpmbuild/BUILD/LimeSuite-20.01.0/src -I/home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/API -I/home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/lime -I/home/abuild/rpmbuild/BUILD/Lime
Suite-20.01.0/src/GFIR -I/home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/protocols -I/home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/ConnectionRegistry -I/home/abuild/rpmbuild/BUILD/Lime
Suite-20.01.0/src/lms7002m_mcu -I/home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/ADF4002 -I/home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/Si5351C -I/home/abuild/rpmbuild/BUILD/LimeSuite
-20.01.0/src/lms7002m -I/home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/FPGA_common -I/home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/external/cpp-feather-ini-parser -I/home/abuild/rpmbuild/
BUILD/LimeSuite-20.01.0/external/kissFFT -I/home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/limeRFE -I/usr/include/libusb-1.0 -I/home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/lms7002_wxg
ui -I/home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/LMS_Programing -I/home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/utilities_gui -I/home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/RF
Spark -I/home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/FPGAcontrols_wxgui -I/home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/numericSlider -I/home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0
/src/fftviewer_wxgui -I/home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/lms7suiteEvents -I/home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/boards_wxgui -isystem /usr/lib64/wx/include/gtk2-
unicode-3.0 -isystem /usr/include/wx-3.0  -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -fl
to=auto -DNDEBUG -fvisibility-inlines-hidden -pthread -O2 -g -DNDEBUG -fPIC   -mmmx -msse -msse2 -msse3 -mfpmath=both -Wall -Wpedantic -fvisibility=hidden -std=gnu++11 -o CMakeFiles/LimeSuit
e.dir/limeRFE/limeRFE_api.cpp.o -c /home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/limeRFE/limeRFE_api.cpp
[   35s] /home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/limeRFE/limeRFE_api.cpp: In function 'rfe_dev_t* RFE_Open(const char*, lms_device_t*)':
[   35s] /home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/limeRFE/limeRFE_api.cpp:14:5: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
[   35s]    14 |     if (dev == nullptr && serialport == nullptr)
[   35s]       |     ^~
[   35s] /home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/limeRFE/limeRFE_api.cpp:17:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
[   35s]    17 |  int result;
[   35s]       |  ^~~
[   35s] /home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/limeRFE/limeRFE_api.cpp: In function 'int RFE_AssignSDRChannels(rfe_dev_t*, int, int)':
[   35s] /home/abuild/rpmbuild/BUILD/LimeSuite-20.01.0/src/limeRFE/limeRFE_api.cpp:179:21: error: control reaches end of non-void function [-Werror=return-type]
[   35s]   179 |     dev->SetChannels(rx, tx);
[   35s]       |     ~~~~~~~~~~~~~~~~^~~~~~~~
[   35s] cc1plus: some warnings being treated as errors
`````

IgnasJarusevicius, 2020-01-29:
I have made a commit c931854ead8130 to remove some warnings, I think it should fix your problem.

mnhauke, 2020-01-29:
https://github.com/myriadrf/LimeSuite/commit/c931854ead81307206bce750c17c2301810b5545 fixes the problem, Thanks.
An updated openSUSE package is on the way...
https://build.opensuse.org/request/show/768275

closed 2020-02-24 by IgnasJarusevicius