myriadrf/LimeSuite #298: An Error in the LimeSuite Libraries

righthalfplane, 2020-02-18:
There appears to be an error in the LimeSuite Libraries. The call to “LMS_StopStream” should stop the transmission - it doesn’t. Even after the call to “LMS_DestroyStream” - the LimeSDR mini continues to transmit. It is not until the call to “LMS_Close” that completely shuts things down that the transmission stop. I have include a version of basicTX.cpp that waits after the calls to LMS_StopStream and LMS_DestroyStream. The video

https://youtu.be/9ZnGq_jK4w0

shows the problem - the LimeSDR mini continues to transmit after the calls to shut the stream down. It happens on MacOS, Ubuntu 18.04, Windows 10 and the Raspberry PI. I down loaded and installed the latest version of the libraries and it still happens.

[basicTX.cpp.zip](https://github.com/myriadrf/LimeSuite/files/4217196/basicTX.cpp.zip)

IgnasJarusevicius, 2020-02-24:
I was pretty sure that LMS_StopStream() works correctly, however I have checked it anyway. And it correctly stops Tx thread, so no data is being send to the board after LMS_StopStream() and FPGA should set DACs to 0 when there is no data from PC. Of course, there will still be carrier (mixer PLL) signal present until Tx is powered down via LMS_EnableChannel()

righthalfplane, 2020-02-26:
My program actually uses the soapySDR routines - I start a stream with -
"setupStream" and "activateStream"
I try to stop the stream with -
"deactivateStream" and "closeStream"
Apparently deactivateStream or closeStream does not have a call to LMS_EnableChannel, because the carrier is still present after they have been called. SoapSDR does not seem to have an equivalent to LMS_EnableChannel.