myriadrf/LimeSuite #300: Crash in RingFIFO::push_packet when buffer size equal to 0

jzampino, 2020-02-24:
This is the result of calling LMS_SetupStream with an lms_stream_t config with fifosize too small.

Eventually, on a call to StreamChannel::Setup, if the config is assigned a bufferLength that results in a size less than pktSize, then on the call to RingFIFO::Resize at Streamer.cpp:38, the resulting bufSize in RingFIFO:Resize is 0.

As such mBufferSize is set to 0.

Subsequently, when in Streamer::ReceivePacketsLoop, the call to push_packet is hit, the resulting check in push_packet (fifo.h:64):

`if (mElementsFilled >= mBufferSize) //buffer might be full, wait for free slots`

is true, and at fifo.h:66:

`mHead = (mHead + 1) % mBufferSize;//advance to next one`

results in a modulo by 0, causing a crash.

There should probably be some check here to ensure mBufferSize is > 0.

You can get around this by either initializing the fifo size to 0, or following the quick start guide and setting fifo size to 1024*1024 samples.

Unfortunately I've inherited the code I'm working on and am not familiar with the LimeSuite other than what I've outlined above, so I can't provide much more insight than that, but hopefully this points someone in the right direction.

IgnasJarusevicius, 2020-02-26:
I have set minimum FIFO size to 4 packets in 6ca12675c246bae5
Anyway, setting small FIFO size can cause a lot of samples to be lost because of FIFO overflows.

gracid, 2020-08-04:
Should be fixed, closing the issue.

closed 2020-08-04 by gracid