myriadrf/LimeSuite #301: LimeUtil --find cannot find LimeSdrUSB

ArrrNeo, 2020-02-26:
Hi All,

I am on centos and have a rt kernel (hope that is not an issue). I am unable to find anything with LimeUtil --find, even though my lsusb shows up the device. I have copied over relevant info. I installed limeSdrSuite following link:
https://discourse.myriadrf.org/t/building-limesuite-on-centos-7/1406

OS: centos:
Linux ran4 3.10.0-1062.9.1.rt56.1033.el7.x86_64 #1 SMP PREEMPT RT Wed Dec 11 09:59:17 CET 2019 x86_64 x86_64 x86_64 GNU/Linux

LimeUtil --info
######################################################
## LimeSuite information summary
######################################################

Version information:
  Library version:      v20.01.0-g6ca12675
  Build timestamp:      2020-02-26
  Interface version:    v2020.1.0
  Binary interface:     20.01-1

System resources:
  Installation root:    /usr/local
  User home directory:  /home/rawatna
  App data directory:   /home/rawatna/.local/share/LimeSuite
  Config directory:     /home/rawatna/.limesuite
  Image search paths:
     - /home/rawatna/.local/share/LimeSuite/images
     - /usr/local/share/LimeSuite/images

Supported connections:
   * PCIEXillybus

SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-gf722f9ce
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib64/SoapySDR/modules0.8
Module found: /usr/local/lib64/SoapySDR/modules0.8/libLMS7Support.so (20.01.0-6ca12675)
Available factories... lime
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

lsusb
Bus 002 Device 003: ID 1d50:6108 OpenMoko, Inc. Myriad-RF LimeSDR
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 021: ID 413c:2003 Dell Computer Corp. Keyboard
Bus 001 Device 005: ID 04e8:6860 Samsung Electronics Co., Ltd Galaxy series, misc. (MTP mode)
Bus 001 Device 083: ID 04e8:6860 Samsung Electronics Co., Ltd Galaxy series, misc. (MTP mode)
Bus 001 Device 004: ID 0b1f:03e9 Insyde Software Corp.
Bus 001 Device 002: ID 0000:0001
Bus 001 Device 089: ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

cat /etc/udev/rules.d/64-limesuite.rules
SUBSYSTEM=="usb", ATTR{idVendor}=="04b4", ATTR{idProduct}=="8613", SYMLINK+="stream-%k", MODE="666"
SUBSYSTEM=="usb", ATTR{idVendor}=="04b4", ATTR{idProduct}=="00f1", SYMLINK+="stream-%k", MODE="666"
SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="601f", SYMLINK+="stream-%k", MODE="666"
SUBSYSTEM=="usb", ATTR{idVendor}=="1d50", ATTR{idProduct}=="6108", SYMLINK+="stream-%k", MODE="666"
SUBSYSTEM=="xillybus", MODE="666", OPTIONS="last_rule"

SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", MODE="0666", SYMLINK+="serial"

Creating new issue as I cannot find relevant help for centos Installation. please help.

Thanks and Regards
Naveen Rawat

IgnasJarusevicius, 2020-02-28:
LimeUtil does not show FX3 or FT601 under supported connections. That means that it was compiled without libusb. So you either did not install libusb package (on Ubuntu its libusb-1.0-0-dev) before running CMake or CMake failed to detect it.

gracid, 2020-08-04:
Since theres no answer to Ignas post, I assume this is solved.

closed 2020-08-04 by gracid

atatogz, 2021-03-25:
I have libusb-1.0-0-dev installed previously to Install LimeUtils but still LimeUtil --info does not show the FX3 or FT601 options and consequently does not list the connected devides. I test in an older computer with older version of Ubuntu and it shows the devices. Can anyone help me?

atatogz, 2021-03-25:
I have this problem when executing cmake (I am install Limesuite from source since I had problems doing it in the other way):
Cannot generate a safe runtime search path for target singleRX because
  there is a cycle in the constraint graph:

    dir 0 is [/usr/lib/x86_64-linux-gnu]
      dir 1 must precede it due to runtime library [libLimeSuite.so.20.10-1]
      dir 2 must precede it due to runtime library [libusb-1.0.so.0]
    dir 1 is [/home/anxo/aradio/LimeSuite/builddir/src]
    dir 2 is [/usr/local/lib]
      dir 1 must precede it due to runtime library [libLimeSuite.so.20.10-1]

  Some of these libraries may not be found correctly.