myriadrf/LimeSuite #306: Gain not set correctly.

enghammer, 2020-05-05:
When initializing LimeSDR Mini in RTLSDR-Airband:
```
type = "soapysdr";
device_string = "driver=lime";
gain = " PGA=10, LNA=20, TIA=5 ";
sample_rate = 30.0;
centerfreq = 132.0;
```

The logfile returns:

```
SoapySDR: device 'driver=lime': gain 'LNA' set to 18.0 dB
SoapySDR: device 'driver=lime': gain 'PGA' set to 10.0 dB
SoapySDR: device 'driver=lime': gain 'TIA' set to 0.0 dB
SoapySDR: device 'driver=lime' initialized
Rx calibration using INTERNAL loopback
Rx ch.A @  132 MHz, BW: 30 MHz, RF input: LNAW, PGA: 22, LNA: 7, TIA: 1
``` 

Any suggestions, why the gain isn't set correctly?
Thanks a lot.

9600, 2020-05-17:
@gracid could you take a look, please.

gracid, 2020-05-19:
The gain is actually being set correctly. You are getting these results because specific gains can have only specific values:
TIA: (0,9,12)
LNA: (0,3,6,9,12,15,18,21,24,25,26,27,28,29,30)
PGA: [-12, 19]
If you want to set a specific gain value in dB I assume you can just set gain=<value in db>

closed 2020-08-04 by gracid