myriadrf/LimeSuite #307: LimeSDR Mini not working (Error getting usb descriptor)

drtrigon, 2020-05-24:
I have serious issues getting LimeSDR Mini to work. I already tried years ago and now again without real success. By "real success" I meant I was able to use qspectrumanalyzer to do a single shot but that is very slow. I was not able to use LimeSuiteGUI FFT or any other tool.

I try to describe what happens. I run on a Ubuntu 20.04 VirtualBox VM. When I install LimeSuite no matter whether I install from repo or pull from git and compile I have issues described below.

Using the repos:

```
$ sudo apt -y install limesuite soapysdr-tools
$ sudo apt -y install python3-pip
$ pip3 install qspectrumanalyzer
```
then I connect the LimeSDR Mini (VM is set to use USB3):
```
$ dmesg
[...]
[  166.572353] usb 2-1: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[  166.594865] usb 2-1: New USB device found, idVendor=0403, idProduct=601f, bcdDevice= 0.00
[  166.594870] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  166.594871] usb 2-1: Product: LimeSDR Mini
[  166.594873] usb 2-1: Manufacturer: Lime Micro
[  166.594874] usb 2-1: SerialNumber: 1D4233B5BE4551
```
now let's check the communication:
```
$ LimeUtil --find
Error getting usb descriptor
failed to get serial number
  * [, media=USB 3.0, module=FT601, addr=24607:1027]
```
this is where issues start.

Intrestingly update does work some times (some times not):
```
$ sudo LimeUtil --update
```
but also throws a lot of these
```
Error getting usb descriptor
failed to get serial number
```

Next run some tests:
```
$ LimeQuickTest --no-gui
```
which passes throwing the same messages (except the LimeSDR Mini was running for some time, then the loopback test fails - I suspect due to high temperature of the board)

Then another connection test:
```
$ SoapySDRUtil --find

[...]
Found device 3
  addr = 24607:1027
  driver = lime
  label =  [USB 3.0]
  media = USB 3.0
  module = FT601
```

Now I can use qspectrumanalyzer/sopay_power by using these settings:
```
File > Settings > Executable: /home/osbox/.local/bin/soapy_power
File > Settings > Device: driver=lime, label= [USB 3.0]
```
that allows me to run a Single shot but is (very) slow - takes 10s or so to show a result.

LimeSuiteGUI is not able to connect to LimeSDR Mini and not able to do anything, it just throws:
```
[11:12:00] INFO: Disconnected control port
[11:12:06] ERROR: Error getting usb descriptor
[11:12:06] ERROR: failed to get serial number
[11:12:13] ERROR: Error getting usb descriptor
[11:12:13] ERROR: failed to get serial number
[11:12:13] ERROR: Error getting usb descriptor
[11:12:13] ERROR: failed to get serial number
[11:12:13] ERROR: failed to get serial number
[11:12:13] INFO: Reference clock 40.00 MHz
[11:12:17] ERROR: TransferPacket: Write failed (ret=0)
[...]
[11:12:17] ERROR: TransferPacket: Write failed (ret=0)
[11:12:17] INFO: Connected Control port: UNKNOWN FW:0 HW:0 Protocol:0 GW:0.0 Ref Clk: 40.00 MHz
```

From here I would like to use other apps that allow me to take power spectra (e.g. sdrangel or similar) but I am not able to make it work.

I must say I bought the LimeSDR Mini years ago and had issues make it run, back then I thought the driver support etc. will improve in the future, but that is now several years ago and still I am not able to bringt it to a usable state. I understand that this is more low-end hardware than e.g. the LimeSDR but still it is not cheap and should work. I really like the idea of having an affordable RF tool, but currently I am getting a bit disapointed and am close to give it up (and trash the mini not to make me throw away even more time trying to make it work).

Can you help me please? That would be really appreciated!

Is it possible that I got buggy hardware?

9600, 2020-05-24:
Firstly, I'd try using it natively and not via a VM. This may not be the cause of your problems, but best to rule this out. If you are running Windows you could try SDR Console, which is generally pretty reliable.

drtrigon, 2020-05-24:
Thanks a lot for your answer!

Running it natively is a good point - came also to my mind after my post - here my findings:

**Summary: this works somehow (so looks like the hardware is ok)**

I used MX Linux (debian derivate) for testing as the win boxes I have access to are not mine, so I don't want to install stuff to them (like drivers that will be needed - as I assume).

I followed exactly the procedure written down above. In this case 
```
$ sudo LimeUtil --find
```
worked as expected not throwing any error or warning messages but the serial number (yeah!). But soapy seems to have issues as:
```
$ sudo SoapySDRUtil --find
```
did not find anything and thus I was not able to use qspectrumanalyzer at all (sigh). But
```
$ sudo LimeSuiteGUI
```
and its FFT module do both work (yeah!).

As I want to get a spectrum I looked for other software and found SigDigger which has an AppImage: https://github.com/BatchDrake/SigDigger/releases

- sdrangel can only be installed by compiling and this looks like a lot of work as snapd does not work with my init system and .deb depends on libavcodec57 - MX Linux details
- sdrangelove does not work as it seems not to support LimeSDR
- sdrconsole seems to be for windows only

So my questions now are:
1. what is another good/powerful but fairly simple to install spectrum analyzer software, you can recommend?
2. how can I get it to work within my VirtualBox VM? (actually this 2nd test was on another hardware, I should move the VM to this hardware and test again...)
3. why does SoapySDRUtil not work in this 2nd test?

9600, 2020-05-25:
For SoapySDR to work you need the LMS7 module installed. On my Ubuntu 18.04 system it is soapysdr-module-lms7. If you build from source you must have SoapySDR installed before building Lime Suite in order for this module to be built.

You shouldn't need to use sudo. Make sure you have the udev rules installed. The package is limesuite-udev. Building from source there is a script to run in the repo.

The issue tracker is for bugs etc. and these are all fairly standard support questions. Closing this issue and if you could please use the [forums](https://discourse.myriadrf.org).

closed 2020-05-25 by 9600

drtrigon, 2020-05-25:
Thanks for your support! Yes it's fine I'll go and check the forums.

Though I had some strange issue with LimeSuiteGUI FFTviewer that did not work until I did some other stuff like using cubicsdr etc. Afterwards it worked again...

Meanwhile I can answer some of my questions:
1. according to https://discourse.myriadrf.org/t/limesdr-mini-gqrx/4537/4 at the time of writing this use sdrangel (linux+win), cubicsdr (linux) or sdrconsole (win) - I go for cubicsdr as in sdrangel I cannot set the center frequency - may be I am just not using the GUI correctly
2. (nothing to add)
3. yes you are right, `soapysdr-module-lms7` needs to be installed also - I though it is a dependency of `soapysdr-tools`