myriadrf/LimeSuite #310: LimeUtil prints wrong serial number

sw, 2020-06-03:
When using `LimeUtil.exe --make=...`, it seems that it always prints the serial number of the first device (0x1d4257ca11494a in this case):
```
C:\Windows\system32>c:\Work\TAS\lime\LimeUtil.exe --find
  * [LimeSDR Mini, media=USB 3, module=FT601, serial=1D4257CA11494A, index=0]
  * [LimeSDR Mini, media=USB 3, module=FT601, serial=1D4257C85DC28D, index=1]
  * [LimeSDR Mini, media=USB 3, module=FT601, serial=1D497878A93785, index=2]
  * [, media=USB, module=FT601, index=3]
  * [LimeSDR Mini, media=USB 3, module=FT601, serial=1D425989E31024, index=4]
  * [LimeSDR Mini, media=USB 3, module=FT601, serial=1D4240C467004C, index=5]
  * [, media=USB, module=FT601, index=6]

C:\Windows\system32>c:\Work\TAS\lime\LimeUtil.exe --make="serial=1D497878A93785"
Make device serial=1D497878A93785
  Device name: LimeSDR-Mini
  Expansion name: UNSUPPORTED
  Firmware version: 6
  Hardware version: 2
  Protocol version: 1
  Gateware version: 1
  Gateware revision: 30
  Gateware target: LimeSDR-Mini
  Serial number: 0x1d4257ca11494a
  Free connection... OK

C:\Windows\system32>c:\Work\TAS\lime\LimeUtil.exe --make="serial=1D4240C467004C"
Make device serial=1D4240C467004C
  Device name: LimeSDR-Mini
  Expansion name: UNSUPPORTED
  Firmware version: 6
  Hardware version: 2
  Protocol version: 1
  Gateware version: 1
  Gateware revision: 30
  Gateware target: LimeSDR-Mini
  Serial number: 0x1d4257ca11494a
  Free connection... OK

C:\Windows\system32>c:\Work\TAS\lime\LimeUtil.exe --info
######################################################
## LimeSuite information summary
######################################################

Version information:
  Library version:      v19.04.1-PothosSDR-2020.01.26-vc14-x64
  Build timestamp:      2020-01-26
  Interface version:    v2019.1.0
  Binary interface:     19.04-1
```
Note that devices with indices 3 and 6 where in use when this was done, thus the missing name.

sw, 2020-06-03:
It seems that adding `module=foo` (literally `foo`!) fixes the problem.

Why is the module even needed and is `foo` the proper value?

In any case the tool should probably warn instead of just connecting to a different device than the one requested.

gracid, 2020-08-04:
Should be fixed, closing the issue.

closed 2020-08-04 by gracid