myriadrf/LimeSuite #313: LimeSuite 20.07 dpkg fails to build on almost all Debian  / Ubuntu

laf0rge, 2020-07-30:
At Osmocom, we are doing automatic builds of LimeSuite in our package feeds.

Since 20.07 was tagged, have consistent build failures on, Debian 10, Debian 8, Debian 9, Ubuntu 16.04, 18.04, 18.10, 19.04, 19.10 - whether for i386, x86_64 or ARM.

They all seem to center around:
<pre>
[  173s] dpkg-shlibdeps: error: couldn't find library libLimeSuite.so.20.07-1 needed by debian/limesuite/usr/bin/LimeSuiteGUI (ELF format: 'elf32-i386'; RPATH: '')
[  175s] dpkg-shlibdeps: error: couldn't find library libLimeSuite.so.20.07-1 needed by debian/limesuite/usr/bin/LimeQuickTest (ELF format: 'elf32-i386'; RPATH: '')
[  176s] dpkg-shlibdeps: error: couldn't find library libLimeSuite.so.20.07-1 needed by debian/limesuite/usr/bin/LimeUtil (ELF format: 'elf32-i386'; RPATH: '')
</pre>

See https://build.opensuse.org/package/live_build_log/network:osmocom:nightly/limesuite/xUbuntu_16.04/i586 https://build.opensuse.org/package/live_build_log/network:osmocom:nightly/limesuite/xUbuntu_16.04/x86_64 https://build.opensuse.org/package/live_build_log/network:osmocom:nightly/limesuite/xUbuntu_18.10/i586 for some examples.

9600, 2020-07-30:
Thanks for alerting us to this, @laf0rge.

laf0rge, 2020-07-30:
The fix might to be in 7990b4a5e577a251ec2374ca663c9c2ee1005a81, but the tag of v20.07.0 still points to one commit before.  So I suggest to quickly tag a v20.07.1 or to remove the v20.07.0 tag.  Otherwise, anyone automatically building the last tag in your repository will be subject to massive build failures.

gracid, 2020-07-30:
Created the new tag pointing to current stable, will update binaries tomorrow.

FFY00, 2020-07-30:
Please do not remove the current tag, never do that, it breaks things for downstream packagers. Creating a new tag is fine.

> Otherwise, anyone automatically building the last tag in your repository will be subject to massive build failures.

Well, anyone building the Debian package :grinning:

gracid, 2020-08-04:
Tag has been changed and I assume it builds correctly.

closed 2020-08-04 by gracid

laf0rge, 2020-08-05:
Thanks @graid.  But the way how it is solved now is also suboptimal.  As you simply tagged the v20.07.1 in the existing git tree, *without* updating the debian/changelog, you end up with the following:

* the package is called v20.7.0
* the contents is v20.7.1

I can live with that for Osmocom, but for users it may be confusing. Geting this right areIMHO basics of dpkg packaging and release manageent processes.  Also, it's relatively easy to have automatic testing in place as part of your CI/CD pipelnine to prevent this from happening.

laf0rge, 2020-08-05:
The good news is that this tag now build on (at least) Debian 9 + 10 + Unstable as well as Ubuntu 18.10, 10.04, 19.10 and 20.04. Still waiting for the Raspbian 10 build results.