myriadrf/LimeSuite #331: Build problem at src/FPGAcontrols_wxgui/FPGAcontrols_wxgui.cpp

cederom, 2020-12-21:
Hello world :-) 

I am porting LimeSuite to FreeBSD. I have encountered this compilation problem. Am I the only one? Should this string be casted anyhow? I am using WX3.1. Any hints welcome :-)

```
/usr/bin/c++ -DLIMERFE -DLIME_DLL -DWIN32_LEAN_AND_MEAN -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D_THREAD_SAFE -D__WXGTK__ -D__unix__=1 -I/u[276/1016]
nclude -I/usr/ports/comms/limesuite/work/LimeSuite-20.10.0/src/lms7002_wxgui -I/usr/ports/comms/limesuite/work/LimeSuite-20.10.0/src/LMS_Programin
g -I/usr/ports/comms/limesuite/work/LimeSuite-20.10.0/src/utilities_gui -I/usr/ports/comms/limesuite/work/LimeSuite-20.10.0/src/RFSpark -I/usr/por
ts/comms/limesuite/work/LimeSuite-20.10.0/src/FPGAcontrols_wxgui -I/usr/ports/comms/limesuite/work/LimeSuite-20.10.0/src/numericSlider -I/usr/port
s/comms/limesuite/work/LimeSuite-20.10.0/src/fftviewer_wxgui -I/usr/ports/comms/limesuite/work/LimeSuite-20.10.0/src/lms7suiteEvents -I/usr/ports/
comms/limesuite/work/LimeSuite-20.10.0/src/boards_wxgui -I/usr/ports/comms/limesuite/work/LimeSuite-20.10.0/src/limeRFE -I/usr/ports/comms/limesui
te/work/LimeSuite-20.10.0/src -I/usr/ports/comms/limesuite/work/LimeSuite-20.10.0/src/API -I/usr/ports/comms/limesuite/work/LimeSuite-20.10.0/src/
lime -I/usr/ports/comms/limesuite/work/LimeSuite-20.10.0/src/GFIR -I/usr/ports/comms/limesuite/work/LimeSuite-20.10.0/src/protocols -I/usr/ports/c
omms/limesuite/work/LimeSuite-20.10.0/src/ConnectionRegistry -I/usr/ports/comms/limesuite/work/LimeSuite-20.10.0/src/lms7002m_mcu -I/usr/ports/com
ms/limesuite/work/LimeSuite-20.10.0/src/ADF4002 -I/usr/ports/comms/limesuite/work/LimeSuite-20.10.0/src/Si5351C -I/usr/ports/comms/limesuite/work/
LimeSuite-20.10.0/src/lms7002m -I/usr/ports/comms/limesuite/work/LimeSuite-20.10.0/src/FPGA_common -I/usr/ports/comms/limesuite/work/LimeSuite-20.
10.0/external/cpp-feather-ini-parser -I/usr/ports/comms/limesuite/work/LimeSuite-20.10.0/external/kissFFT -I/usr/ports/comms/limesuite/work/LimeSu
ite-20.10.0/src/threadHelper -I/usr/ports/comms/limesuite/work/LimeSuite-20.10.0/src/oglGraph -isystem /usr/local/lib/wx/include/gtk3-unicode-3.1 
-isystem /usr/local/include/wx-3.1 -O2 -pipe -fstack-protector-strong -fno-strict-aliasing -pthread -O2 -pipe -fstack-protector-strong -fno-strict
-aliasing -pthread -std=gnu++11 -MD -MT src/CMakeFiles/LimeSuiteGUI.dir/FPGAcontrols_wxgui/FPGAcontrols_wxgui.cpp.o -MF src/CMakeFiles/LimeSuiteGU
I.dir/FPGAcontrols_wxgui/FPGAcontrols_wxgui.cpp.o.d -o src/CMakeFiles/LimeSuiteGUI.dir/FPGAcontrols_wxgui/FPGAcontrols_wxgui.cpp.o -c /usr/ports/c
omms/limesuite/work/LimeSuite-20.10.0/src/FPGAcontrols_wxgui/FPGAcontrols_wxgui.cpp
/usr/ports/comms/limesuite/work/LimeSuite-20.10.0/src/FPGAcontrols_wxgui/FPGAcontrols_wxgui.cpp:169:71: error: cannot pass object of non-trivial t
ype 'const wxString' through variadic function; call will abort at runtime [-Wnon-pod-varargs]
        lime::debug("WFM reading: Input file can not be opened (%s)", filename);
                                                                      ^
1 error generated.                  
```

cederom, 2020-12-21:
Fix proposition here https://github.com/myriadrf/LimeSuite/pull/332:
```
lime::debug("WFM reading: Input file can not be opened (%s)", filename.mb_str().data());
```

rjonaitis, 2021-01-27:
Fix merged

closed 2021-01-27 by rjonaitis