myriadrf/LimeSuite #338: wx assertions on FreeBSD

cederom, 2020-12-28:
Hello world, we observe lots of assertions on startup on FreeBSD. Any hints on the cause and how to fix are welcome :-)

[image]

rjonaitis, 2021-01-27:
Most likely caused by mixing of wxEXPAND and wxALIGN flags in forms creation. Like: `StaticBoxSizer1->Add(btnLoadFile, 1, wxEXPAND | wxALIGN_CENTER_HORIZONTAL | wxALIGN_CENTER_VERTICAL, 5);`
The backtrace should contain the stacktrace of which UI elements cause that.
Most of the UI is generated code by wxFormBuilder, maybe it should be enough just to regenerate it with newer version of it.

cederom, 2021-01-27:
Hello @rjonaitis and thank you for feedback and hints! Below is the backtrace. Please let me know if you need additional information. It looks like wx changed api a bit between 2 and 3 and maybe its time to update sources to match 3 :-)
```
ASSERT INFO:
./src/common/sizer.cpp(1492): assert "!(flags & (wxALIGN_BOTTOM | wxALIGN_CENTRE_VERTICAL)) || !(flags & (wxALIGN_RIGHT | wxALIGN_CENTRE_HORIZONTAL))" failed in DoInsert(): wxEXPAND flag will be overridden by alignment flags

BACKTRACE:
[1] std::__1::char_traits<char>::length(char const*)
[2] void std::__1::allocator<unsigned char>::construct<unsigned char, unsigned char const&>(unsigned char*, unsigned char const&)
[3] std::__1::char_traits<char>::eof()
[4] wxNavigationEnabled<wxTopLevelWindow>::RemoveChild(wxWindowBase*)
[5] std::__1::char_traits<char>::eof()
[6] _ZNSt3__116allocator_traitsINS_9allocatorIdEEE20__construct_backwardIdEENS_9enable_ifIXaaooL_ZNS_17integral_constantIbLb1EE5valueEEntsr15__has_constructIS2_PT_S8_EE5valuesr31is_trivially_move_constructibleIS8_EE5valueEvE4typeERS2_S9_S9_RS9_
[7] void std::__1::allocator<unsigned char>::construct<unsigned char, unsigned char const&>(unsigned char*, unsigned char const&)
[8] void std::__1::allocator_traits<std::__1::allocator<lime::ConnectionHandle> >::__destroy<lime::ConnectionHandle>(std::__1::integral_constant<bool, true>, std::__1::allocator<lime::ConnectionHandle>&, lime::ConnectionHandle*)
[9] void std::__1::allocator<unsigned char>::construct<unsigned char, unsigned char const&>(unsigned char*, unsigned char const&)
[10] void std::__1::allocator<unsigned char>::construct<unsigned char, unsigned char const&>(unsigned char*, unsigned char const&)
[11] void std::__1::allocator<unsigned char>::construct<unsigned char, unsigned char const&>(unsigned char*, unsigned char const&)
```

rjonaitis, 2021-11-06:
These asserts are only triggered when program is linked to debug libraries of wxWidgets. Linking to release version of wxWidgets will get rid of them.

cederom, 2021-11-06:
Thank you for your time @rjonaitis :-)

Looks like we have debug enabled by default on FreeBSD packages because I am using generic packages :-)

Is this debug switch part of the LimeSuite build?

Why should not we fix the source of the assert anyway instead of just hiding them? :-)

rjonaitis, 2021-11-07:
LimeSuite is already choosing to use wxWidgets without debug, but I guess if the system provided libraries do not have that version then it fallbacks to whatever is available (or maybe the wxWidgets libraries were compiled with forced debug enabled). https://github.com/myriadrf/LimeSuite/blob/966b1076b2a03251212f062b5776226ece4bd3e8/src/CMakeLists.txt#L176

Yes, these assertions should be fixed, but it might take a while, because that requires going through basically every single UI element and modifying their flags.

rjonaitis, 2022-04-07:
Assertions and compatibility with wxWidgets 3.1.5 has been fixed.

closed 2022-04-07 by rjonaitis

cederom, 2022-04-07:
TANK U SIR! =)

cederom, 2022-04-07:
I can confirm that assertions are gone in the upstream :-) Time for a new release? :-)

cederom, 2022-09-06:
ping @rjonaitis any chance for a new release? last one was almost 2 years ago and there are some important fixes in the upstream :-)

rjonaitis, 2022-09-08:
tagged new release

cederom, 2022-09-08:
TANK U SIR! =)