myriadrf/LimeSuite #339: LimeSDR-MINI LimeQuickTest Self-Test FAILED on FreeBSD

cederom, 2020-12-29:
@JanBeh's unit (new unit with metal casing) was at the latest firmware verison (1.30 or 30) and sometimes fails one of the RF loopback tests as follows. What OS do you run this `LimeQuickTest` @JanBeh - is it Linux or FreeBSD?

    % ./LimeQuickTest
    [ TESTING STARTED ]
    ->Start time: Thu Dec 24 14:28:52 2020
    ->LimeSuite version: 20.10.0-unknown
    
    ->Device: LimeSDR Mini, media=USB 3.0, module=FT601, addr=24607:1027, serial=1D589769035A75, HW=3, GW=1.30
      Serial Number: 1D589769035A75
     Chip temperature: 45 C
    
    [ Clock Network Test ]
    ->REF clock test
      Test results: 54133; 1794; 14991 - PASSED
    ->VCTCXO test
      Results : 6711044 (min); 6711207 (max) - PASSED
    ->Clock Network Test PASSED
    
    [ FPGA EEPROM Test ]
    ->Read EEPROM
    ->Read data: 13 0A 17 13 0A 17 02
    ->FPGA EEPROM Test PASSED
    
    [ LMS7002M Test ]
    ->Perform Registers Test
    ->External Reset line test
      Reg 0x20: Write value 0xFFFD, Read value 0xFFFD
      Reg 0x20: value after reset 0x0FFFF
    ->LMS7002M Test PASSED
    
    [ RF Loopback Test ]
    ->Configure LMS
    ->Run Tests (TX_2 -> LNA_W):
      CH0 (SXR=1000.0MHz, SXT=1005.0MHz): Result:(-20.7 dBFS, 5.00 MHz) - FAILED
    ->Run Tests (TX_1 -> LNA_H):
      CH0 (SXR=2100.0MHz, SXT=2105.0MHz): Result:(-19.5 dBFS, 5.00 MHz) - PASSED
    ->RF Loopback Test FAILED
    
    => Board tests FAILED <=
    
    Elapsed time: 2.30 seconds

The same problem seems to occur on @cederom device:

```
# LimeQuickTest

[ TESTING STARTED ]
->Start time: Tue Dec 29 01:57:42 2020
->LimeSuite version: 20.10.0-unknown

Warning: Multiple devices detected, connecting to the first device
->Device: LimeSDR Mini, media=USB 3.0, module=FT601, addr=24607:1027, serial=1D5897C7837267, HW=3, GW=1.30
  Serial Number: 1D5897C7837267
 Chip temperature: 62 C

[ Clock Network Test ]
->REF clock test
  Test results: 61409; 9070; 22267 - PASSED
->VCTCXO test
  Results : 6711061 (min); 6711222 (max) - PASSED
->Clock Network Test PASSED

[ FPGA EEPROM Test ]
->Read EEPROM
->Read data: 13 0A 17 13 0A 17 02
->FPGA EEPROM Test PASSED

[ LMS7002M Test ]
->Perform Registers Test
->External Reset line test
  Reg 0x20: Write value 0xFFFD, Read value 0xFFFD
  Reg 0x20: value after reset 0x0FFFF
->LMS7002M Test PASSED

[ RF Loopback Test ]
->Configure LMS
CGEN: Freq=61.44 MHz, VCO=2.33472 GHz, INT=57, FRAC=385875, DIV_OUTCH_CGEN=18
CGEN: Freq=61.44 MHz, VCO=2.33472 GHz, INT=57, FRAC=385875, DIV_OUTCH_CGEN=18
CGEN: Freq=61.44 MHz, VCO=2.33472 GHz, INT=57, FRAC=385875, DIV_OUTCH_CGEN=18
CGEN: Freq=122.88 MHz, VCO=2.21184 GHz, INT=54, FRAC=310378, DIV_OUTCH_CGEN=8
CGEN: Freq=122.88 MHz, VCO=2.21184 GHz, INT=54, FRAC=310378, DIV_OUTCH_CGEN=8
->Run Tests (TX_2 -> LNA_W):
  CH0 (SXR=1000.0MHz, SXT=1005.0MHz): Result:(-20.9 dBFS, 5.00 MHz) - FAILED
->Run Tests (TX_1 -> LNA_H):
  CH0 (SXR=2100.0MHz, SXT=2105.0MHz): Result:(-21.0 dBFS, 5.00 MHz) - FAILED
->RF Loopback Test FAILED

=> Board tests FAILED <=

Elapsed time: 3.28 seconds

# uname -a
FreeBSD 0xCFMX4 12.2-RELEASE-p1 FreeBSD 12.2-RELEASE-p1 GENERIC  amd64
```

Problems to solve:
* `LimeQuickTest` needs documentation improvement [1] - it is unclear what test are performed, why, and what is their outcome. Also no information if test envronment needs some specific setup in order for some tests to pass.
* Should testing stop at first fail or continue with other tests? Performing all tests even some of them can fail would produce more detailed and useful results.
* Does `LimeQuickTest` depend on OS and/or Firmware? May some tests fail because of OS issues or the old firmware missing some new functionalities?

[1] https://wiki.myriadrf.org/LimeQuickTest

JanBeh, 2020-12-29:
> What OS do you run this `LimeQuickTest` @JanBeh - is it Linux or FreeBSD?

I did the test on FreeBSD. I don't know yet if the problem is FreeBSD specific though, or if it happens on Linux as well.

cederom, 2020-12-29:
Thanks @JanBeh :-) It looks like 4 errors may be related to firmware issues:
1.  LibUSB open / reset impossible on FreeBSD when working as user (only root can reset hardware USB ports).
2. Hardware Self-Test fail with LimeQuickTest.
3. MCU error 3 / Tx Calibration: MCU error 3 (SXR tune failed).
4. MCU error 5 / Tx Calibration: MCU error 5 (Loopback signal weak: not connected/insufficient gain?).

I have reported them on LimeSDR-Mini github repo https://github.com/myriadrf/LimeSDR-Mini/issues/2 but it got closed ultra fast and I was redirected back here by @9600 :-)

The questions are:
1. Is LimeSDR-Mini firmware Open-Source?
2. Where is LimeSDR-Mini firmware source code located?
3. How to fix (mentioned) issues and improve LimeSDR-Mini firmware so it works with no problem on all platforms.

daniestevez, 2020-12-29:
For your questions 1 and 2, depending on what you mean by "firmware", the answer might be in [LimeSDR-Mini_GW](https://github.com/myriadrf/LimeSDR-Mini_GW), which contains the VHDL project for the FPGA image, or in the [mcu_program folder in LimeSuite](https://github.com/myriadrf/LimeSuite/tree/master/mcu_program), which I think that contains the source for the programs that run on the LMS7002M MCU (but I'm not so familiar with it).

9600, 2020-12-29:
> I have reported them on LimeSDR-Mini github repo myriadrf/LimeSDR-Mini#2 but it got closed ultra fast and I was redirected back here by @9600 :-)

If you have the latest gateware/firmware loaded it should not be some issue related to this and as noted, would imagine it is either a problem with the SDR hardware, else using it (or probably rather Lime Suite) with FreeBSD.

In addition to the gateware there are two firmwares: one for the NIOS II soft processor implemented in the FPGA, and another for the MCU in the LMS7002M RFIC. The former configures the transceiver etc. and implements LMS64C protocol for control over USB, while the latter runs calibration routines on the RFIC. I doubt the issue is with either, provided that you have the latest Lime Suite (and hence RFIC MCU firmware) and LimeSDR Mini gateware (and hence NIOS II firmware).

The best place to get support debugging potential board issues is [the forums](https://discourse.myriadrf.org/).

cederom, 2020-12-30:
Thanks @9600 :-) I am new to LimeSDR/LimeSuite thus my questions :-) There are some problems related with FreeBSD but some are platform independent. I will double check on Linux and Windows with the same hardware just to make sure :-) I got hardware directly from MyriadRF so I guess it should be okay. I am using newest update and just to make sure I did `LimeUtil --update --force`. I also will try the forum, thanks, I just got used to solve problems here at source + provide patches (see pull requests that I already created), but there may be some useful hints over there :-)

Our biggest problem right now is LibUSB reset - on FreeBSD a non-root user cannot really reset a hardware. Do you think a device could self-reset? Is this reset really necessary?

cederom, 2020-12-30:
Tests passed on Windows. 

[image]

```
[ TESTING STARTED ]
->Start time: Wed Dec 30 13:25:35 2020
->LimeSuite version: 20.10.0-g1480bfea

->Device: LimeSDR Mini, media=USB 3, module=FT601, serial=1D5897C7837267, index=0, HW=3, GW=1.30
  Serial Number: 1D5897C7837267
 Chip temperature: 29 C

[ Clock Network Test ]
->REF clock test
  Test results: 3949; 17146; 30343 - PASSED
->VCTCXO test
  Results : 6711015 (min); 6711179 (max) - PASSED
->Clock Network Test PASSED

[ FPGA EEPROM Test ]
->Read EEPROM
->Read data: 13 0A 17 13 0A 17 02
->FPGA EEPROM Test PASSED

[ LMS7002M Test ]
->Perform Registers Test
->External Reset line test
  Reg 0x20: Write value 0xFFFD, Read value 0xFFFD
  Reg 0x20: value after reset 0x0FFFF
->LMS7002M Test PASSED

[ RF Loopback Test ]
->Configure LMS
->Run Tests (TX_2 -> LNA_W):
  CH0 (SXR=1000.0MHz, SXT=1005.0MHz): Result:(-19.2 dBFS, 5.00 MHz) - PASSED
->Run Tests (TX_1 -> LNA_H):
  CH0 (SXR=2100.0MHz, SXT=2105.0MHz): Result:(-19.1 dBFS, 5.00 MHz) - PASSED
->RF Loopback Test PASSED

=> Board tests PASSED <=

Elapsed time: 6.27 seconds
```

9600, 2020-12-30:
Actually, looking back at those earlier failures, if it was just with e.g.:

> Result:(-20.7 dBFS, 5.00 MHz)

A loopback test with results just very slightly out of spec and just a dB or two lower than when it passes, I don't think there is anything to worry about and any differences between this and testing under Linux/Windows must be either coincidental or involve other external factors, such as the board was a bit warmer when the test failed. In terms of communicating with the board and configuring it, everything looks good.

As to USB reset, I don't know, but perhaps @ztamosevicius or someone in his team could advise.

cederom, 2020-12-30:
Thanks @9600 I am calmer now :-) Shall we use some sort of rf cable on rx-tx line to make sure test environment is correct? Also the temperature seems to impact test results - all around above 45C seems to fail? Shall we change test-pass criteria in `LimeQuickTest`, or introduce `WARNING` level instead `FAILED`? :-)

@JanBeh can you please retry your tests with cold device (below 45C) and antennas disconnected to confirm this issue was caused by board temperature above 45C?

9600, 2020-12-30:
> Shall we use some sort of rf cable on rx-tx line to make sure test environment is correct? 

It is important to have absolutely nothing connected to RF ports when running LimeQuickTest, as even a short cable can affect the results. Loopback is done via RF switches on the board.

> Also the temperature seems to impact test results - all around above 45C seems to fail? 

The performance will degrade as temperature increases and the pass/fail criteria are set for testing the board when cold, hence if you test when it is not, the results are not valid. So when it passes at colder temperatures, a failure when the board is warmer is not a hard failure and rather instead an invalid test because you've not created the correct test environment.

> Shall we change test-pass criteria in LimeQuickTest, or introduce WARNING level instead FAILED? :-)

We don't propose to make any changes to the way that the test operates or the results it provides, and rather instead we would request that people follow the test instructions provided :o)

cederom, 2020-12-30:
> We don't propose to make any changes to the way that the test operates or the results it provides, and rather instead we would request that people follow the test instructions provided :o)

Cool, I guess you really want folks to read wiki and search forums, because none instructions are contained within the software itself thus first contact confusion. A short hint message within the software would really do the job.

Also it is unknown whether testing stops on first failure or all tests are performed completing the test list because no result table is printed at the end (text mode) just `=> Board tests FAILED <=` which gives no clue what tests were (not) performed what failed and what fails are critical. I think there might be some room for improvement :-)

Thank you for your time folks :-)

9600, 2020-12-30:
This is something that's been discussed previously and following which, I had thought text had been added to instruct using a cold board with no RF cabling attached. However, just running LimeQuickTest on my laptop here, I didn't see anything. Let me discuss with the team.

cederom, 2020-12-30:
Just for reference I did some testing on Kali Linux with a default provided binary packages that is LimeSuite 20.01+dfsg-2 that is a bit behind current 20.10.0. There was no failure at all even with antennas connected and hot device :-)

```
user@linux:~$ uname -a
Linux linux 5.6.0-kali2-amd64 #1 SMP Debian 5.6.14-2kali1 (2020-06-10) x86_64 GNU/Linux

user@linux:~$ LimeUtil --find
  * [LimeSDR Mini, media=USB 3.0, module=FT601, addr=24607:1027, serial=1D5897C7837267]

user@linux:~$ LimeUtil --info
######################################################
## LimeSuite information summary
######################################################

Version information:
  Library version:      v20.01.0+dfsg-2
  Build timestamp:      2020-02-04
  Interface version:    v2020.1.0
  Binary interface:     20.01-1

System resources:
  Installation root:    /usr
  User home directory:  /home/user
  App data directory:   /home/user/.local/share/LimeSuite
  Config directory:     /home/user/.limesuite
  Image search paths:
     - /home/user/.local/share/LimeSuite/images
     - /usr/share/LimeSuite/images

Supported connections:
   * FT601
   * FX3
   * PCIEXillybus

user@linux:~$ LimeQuickTest 
[ TESTING STARTED ]
->Start time: Wed Dec 30 20:56:53 2020

USB reset failed
Failed to open device
Failed to open. Device is busy.
Error: Unable to connect
Failed to connect
user@linux:~$ LimeQuickTest 
[ TESTING STARTED ]
->Start time: Wed Dec 30 20:56:56 2020

->Device: LimeSDR Mini, media=USB 3.0, module=FT601, addr=24607:1027, serial=1D5897C7837267
  Serial Number: 1D5897C7837267

[ Clock Network Test ]
->REF clock test
  Test results: 40331; 53528; 1190 - PASSED
->VCTCXO test
  Results : 6710995 (min); 6711160 (max) - PASSED
->Clock Network Test PASSED

[ FPGA EEPROM Test ]
->Read EEPROM
->Read data: 13 0A 17 13 0A 17 02
->FPGA EEPROM Test PASSED

[ LMS7002M Test ]
->Perform Registers Test
->External Reset line test
  Reg 0x20: Write value 0xFFFD, Read value 0xFFFD
  Reg 0x20: value after reset 0x0FFFF
->LMS7002M Test PASSED
                                                                                                                                                                                                                 
[ RF Loopback Test ]                                                                                                                                                                                             
[... 62 lines cut ...]

[ FPGA EEPROM Test ]
->Read EEPROM
->Read data: 13 0A 17 13 0A 17 02
->FPGA EEPROM Test PASSED

[ LMS7002M Test ]
->Perform Registers Test
->External Reset line test
  Reg 0x20: Write value 0xFFFD, Read value 0xFFFD
  Reg 0x20: value after reset 0x0FFFF
->LMS7002M Test PASSED

[ RF Loopback Test ]
->Configure LMS
->Run Tests (TX_2 -> LNA_W):
  CH0 (SXR=1000.0MHz, SXT=1005.0MHz): Result:(-13.9 dBFS, 5.00 MHz) - PASSED
->Run Tests (TX_1 -> LNA_H):
  CH0 (SXR=2100.0MHz, SXT=2105.0MHz): Result:(-12.4 dBFS, 5.00 MHz) - PASSED
->RF Loopback Test PASSED

=> Board tests PASSED <=

Elapsed time: 2.58 seconds
```

cederom, 2020-12-30:
With version 20.10.0 compiled from sources there are some fails and with this version I can see the board temperature. The difference is also in Loopback Test Results: -13.9 dBFS vs -19..20.2 dBFS leading to fails.

Under Linux I can use LimeSuite + LimeSDR-Mini as non-root user, so the LibUSB'reset seems blocker for FreeBSD non-root use.

```
user@linux:~/cederom/work/limesuite/limesuite.git$ LimeUtil --info
######################################################
## LimeSuite information summary
######################################################

Version information:
  Library version:      v20.10.0-g1480bfea
  Build timestamp:      2020-12-30
  Interface version:    v2020.10.0
  Binary interface:     20.10-1

System resources:
  Installation root:    /usr/local
  User home directory:  /home/user
  App data directory:   /home/user/.local/share/LimeSuite
  Config directory:     /home/user/.limesuite
  Image search paths:
     - /home/user/.local/share/LimeSuite/images
     - /usr/local/share/LimeSuite/images

Supported connections:
SetOSThreadPriority: Failed to set priority(3), schec_prio(2), policy(1), ret(50)
SetOSThreadPriority: Failed to set priority(3), schec_prio(2), policy(1), ret(50)
   * FT601
   * FX3
   * PCIEXillybus

user@linux:~/cederom/work/limesuite/limesuite.git$ LimeUtil --find
SetOSThreadPriority: Failed to set priority(3), schec_prio(2), policy(1), ret(50)
SetOSThreadPriority: Failed to set priority(3), schec_prio(2), policy(1), ret(50)
  * [LimeSDR Mini, media=USB 3.0, module=FT601, addr=24607:1027, serial=1D5897C7837267]

user@linux:~/cederom/work/limesuite/limesuite.git$ LimeQuickTest 
[ TESTING STARTED ]
->Start time: Wed Dec 30 21:29:43 2020
->LimeSuite version: 20.10.0-g1480bfea

->Device: LimeSDR Mini, media=USB 3.0, module=FT601, addr=24607:1027, serial=1D5897C7837267, HW=3, GW=1.30
  Serial Number: 1D5897C7837267
 Chip temperature: 52 C

[ Clock Network Test ]
->REF clock test
  Test results: 53004; 660; 13857 - PASSED
->VCTCXO test
  Results : 6711056 (min); 6711218 (max) - PASSED
->Clock Network Test PASSED

[ FPGA EEPROM Test ]
->Read EEPROM
->Read data: 13 0A 17 13 0A 17 02
->FPGA EEPROM Test PASSED

[ LMS7002M Test ]
->Perform Registers Test
->External Reset line test
  Reg 0x20: Write value 0xFFFD, Read value 0xFFFD
  Reg 0x20: value after reset 0x0FFFF
->LMS7002M Test PASSED

[ RF Loopback Test ]
->Configure LMS
CGEN: Freq=61.44 MHz, VCO=2.33472 GHz, INT=57, FRAC=385875, DIV_OUTCH_CGEN=18
CGEN: Freq=61.44 MHz, VCO=2.33472 GHz, INT=57, FRAC=385875, DIV_OUTCH_CGEN=18
CGEN: Freq=61.44 MHz, VCO=2.33472 GHz, INT=57, FRAC=385875, DIV_OUTCH_CGEN=18
CGEN: Freq=122.88 MHz, VCO=2.21184 GHz, INT=54, FRAC=310378, DIV_OUTCH_CGEN=8
CGEN: Freq=122.88 MHz, VCO=2.21184 GHz, INT=54, FRAC=310378, DIV_OUTCH_CGEN=8
->Run Tests (TX_2 -> LNA_W):
  CH0 (SXR=1000.0MHz, SXT=1005.0MHz): Result:(-20.2 dBFS, 5.00 MHz) - FAILED
->Run Tests (TX_1 -> LNA_H):
  CH0 (SXR=2100.0MHz, SXT=2105.0MHz): Result:(-19.3 dBFS, 5.00 MHz) - PASSED
->RF Loopback Test FAILED

=> Board tests FAILED <=

Elapsed time: 2.84 seconds

user@linux:~/cederom/work/limesuite/limesuite.git$ LimeQuickTest 
[ TESTING STARTED ]
->Start time: Wed Dec 30 21:30:35 2020
->LimeSuite version: 20.10.0-g1480bfea

->Device: LimeSDR Mini, media=USB 3.0, module=FT601, addr=24607:1027, serial=1D5897C7837267, HW=3, GW=1.30
  Serial Number: 1D5897C7837267
 Chip temperature: 50 C

[ Clock Network Test ]
->REF clock test
  Test results: 45675; 58870; 6531 - PASSED
->VCTCXO test
  Results : 6711058 (min); 6711219 (max) - PASSED
->Clock Network Test PASSED

[ FPGA EEPROM Test ]
->Read EEPROM
->Read data: 13 0A 17 13 0A 17 02
->FPGA EEPROM Test PASSED

[ LMS7002M Test ]
->Perform Registers Test
->External Reset line test
  Reg 0x20: Write value 0xFFFD, Read value 0xFFFD
  Reg 0x20: value after reset 0x0FFFF
->LMS7002M Test PASSED

[ RF Loopback Test ]
->Configure LMS
CGEN: Freq=61.44 MHz, VCO=2.33472 GHz, INT=57, FRAC=385875, DIV_OUTCH_CGEN=18
CGEN: Freq=61.44 MHz, VCO=2.33472 GHz, INT=57, FRAC=385875, DIV_OUTCH_CGEN=18
CGEN: Freq=61.44 MHz, VCO=2.33472 GHz, INT=57, FRAC=385875, DIV_OUTCH_CGEN=18
CGEN: Freq=122.88 MHz, VCO=2.21184 GHz, INT=54, FRAC=310378, DIV_OUTCH_CGEN=8
CGEN: Freq=122.88 MHz, VCO=2.21184 GHz, INT=54, FRAC=310378, DIV_OUTCH_CGEN=8
->Run Tests (TX_2 -> LNA_W):
  CH0 (SXR=1000.0MHz, SXT=1005.0MHz): Result:(-20.1 dBFS, 5.00 MHz) - FAILED
->Run Tests (TX_1 -> LNA_H):
  CH0 (SXR=2100.0MHz, SXT=2105.0MHz): Result:(-20.1 dBFS, 5.00 MHz) - FAILED
->RF Loopback Test FAILED

=> Board tests FAILED <=

Elapsed time: 2.81 seconds

```

JuliDi, 2021-06-24:
>With version 20.10.0 compiled from sources there are some fails and with this version I can see the board temperature. The difference is also in Loopback Test Results: -13.9 dBFS vs -19..20.2 dBFS leading to fails.

Same thing here on Ubuntu 20.04 with a brand new LimeSDR mini 1.3. All tests pass except for the RF Loopback Test. I get around -21 to -23 dBFS (probably depending on the temperature).

Here with the device being hot:
```
[ RF Loopback Test ]
->Configure LMS
->Run Tests (TX_2 -> LNA_W):
  CH0 (SXR=1000.0MHz, SXT=1005.0MHz): Result:(-22.2 dBFS, 5.00 MHz) - FAILED
->Run Tests (TX_1 -> LNA_H):
  CH0 (SXR=2100.0MHz, SXT=2105.0MHz): Result:(-22.8 dBFS, 5.00 MHz) - FAILED
->RF Loopback Test FAILED
```
or after letting it cool down a few minutes:
```
[ RF Loopback Test ]
->Configure LMS
->Run Tests (TX_2 -> LNA_W):
  CH0 (SXR=1000.0MHz, SXT=1005.0MHz): Result:(-21.4 dBFS, 5.00 MHz) - FAILED
->Run Tests (TX_1 -> LNA_H):
  CH0 (SXR=2100.0MHz, SXT=2105.0MHz): Result:(-21.8 dBFS, 5.00 MHz) - FAILED
->RF Loopback Test FAILED
```

Using Ubuntu 20.04 on a VM, host is on macOS. 
```
$ LimeUtil --info 
######################################################
## LimeSuite information summary
######################################################

Version information:
  Library version:	v20.10.0-g1480bfea
  Build timestamp:	2021-06-20
  Interface version:	v2020.10.0
  Binary interface:	20.10-1

System resources:
  Installation root:	/usr/local
  User home directory:	/home/sdr
  App data directory:	/home/sdr/.local/share/LimeSuite
  Config directory:	/home/sdr/.limesuite
  Image search paths:
     - /home/sdr/.local/share/LimeSuite/images
     - /usr/local/share/LimeSuite/images

Supported connections:
   * FT601
   * FX3
   * PCIEXillybus
```
 
Is this something to worry about/is the device malfunctioning? Or is this just a LimeQuickTest bug? (The device does seem to work in the LimeSuiteGUI and gqrx; is there any way to confirm this?)

ztamosevicius, 2021-06-25:
If loopback test result is more than -40 dBFS, then everything is OK. The current limit if -20 dBFS which is from the past and now we just do not want to change it.