myriadrf/LimeSuite #340: MCU error 3 / SXR tune failed on LimeSDR-Mini

cederom, 2020-12-29:
Issue seems to be reproducible with a [test program](https://sapphire.magnetkern.de/paste_1d1b4528287690e1e2dc1200acb41427/limetest_2020-09-03_0731Z.c). 

Under FreeBSD, ensure that ``libLMS7Support.so`` of Lime Suite has been installed in `/usr/local/lib/SoapySDR/modules0.7` and compile with: ```cc -Wall -o limetest-mcu-error -I /usr/local/include limetest_2020-09-03_0731Z.c -L /usr/local/lib -l SoapySDR```.

Further notes on that issue can be found in the thread MCU error 3 (SXR tune failed) on LimeSDR USB: https://discourse.myriadrf.org/t/mcu-error-3-sxr-tune-failed-on-limesdr-usb/5303.

I can confirm this issue on FreeBSD but it is known to exist among different platforms, so it may be a firmware issue?

```
# ./limetest-mcu-error 
[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D5897C7837267'
[INFO] RemoteControl Listening on port: 5000

[INFO] Reference clock 40.00 MHz
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
CGEN: Freq=61.44 MHz, VCO=2.33472 GHz, INT=57, FRAC=385875, DIV_OUTCH_CGEN=18
CGEN: Freq=61.44 MHz, VCO=2.33472 GHz, INT=57, FRAC=385875, DIV_OUTCH_CGEN=18
CGEN: Freq=61.44 MHz, VCO=2.33472 GHz, INT=57, FRAC=385875, DIV_OUTCH_CGEN=18
[INFO] LMS7002M register cache: Disabled
CGEN: Freq=49.152 MHz, VCO=2.3593 GHz, INT=57, FRAC=1030121, DIV_OUTCH_CGEN=23
CGEN: Freq=49.152 MHz, VCO=2.3593 GHz, INT=57, FRAC=1030121, DIV_OUTCH_CGEN=23
MCU programming : 16384/16384
MCU Programming finished, 2570 ms
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
CGEN: Freq=61.44 MHz, VCO=2.33472 GHz, INT=57, FRAC=385875, DIV_OUTCH_CGEN=18
[INFO] TX LPF configured
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 11.563 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
CGEN: Freq=61.44 MHz, VCO=2.33472 GHz, INT=57, FRAC=385875, DIV_OUTCH_CGEN=18
[INFO] TX LPF configured
Tx calibration using MCU INTERNAL loopback
Tx ch.A @   33 MHz, BW: 11.563 MHz, RF output: BAND2, Gain: 4
Current MCU firmware: 5, DC/IQ calibration full
MCU Ref. clock: 40 MHz
[ERROR] Tx Calibration: MCU error 3 (SXR tune failed)

# uname -a
FreeBSD 0xCFMX4 12.2-RELEASE-p1 FreeBSD 12.2-RELEASE-p1 GENERIC  amd64
```

JanBeh, 2021-01-22:
Thanks to @cederom for opening a ticket on this. I discovered the problem when I intended to send on HF (<30 MHz). The only way I know to send on frequencies below 30 MHz is to set the RF frequency to 30 MHz or higher and to increase the bandwidth to be able to reach to frequencies below 30 MHz. So this issue is an actual blocker if you want to transmit on HF bands.

When the bandwidth is set too small, the device will appear to work but the emitted power is very very low. Setting a wider bandwidth results in the error described above. This effectively makes the LimeSDR-Mini unsuitable for working on HF bands (unless I missed something?). I hope this can be fixed somehow.

cederom, 2021-01-22:
Hey @JanBeh :-) I have noticed that this project prefers to work on MyriadRF Forum [1] please ask our question there :-)

[1] https://discourse.myriadrf.org/

JanBeh, 2021-01-23:
> Hey @JanBeh :-) I have noticed that this project prefers to work on MyriadRF Forum [1] please ask our question there :-)
> 
> [1] https://discourse.myriadrf.org/

I didn't make my post above to ask a question but to provide background information on the issue, i.e. explain the context and impact of the problem. I encountered this problem not during tests but when I tried to transmit on HF bands. It is not just a theoretical issue but a real blocker for me.

Several other people could reproduce my problem [here](https://discourse.myriadrf.org/t/mcu-error-3-sxr-tune-failed-on-limesdr-usb/5303/29?u=dl9jbe), [here](https://discourse.myriadrf.org/t/mcu-error-3-sxr-tune-failed-on-limesdr-usb/5303/32?u=dl9jbe), and also @cederom above.

Regarding my question in parenthesis, I already [did ask there](https://discourse.myriadrf.org/t/mcu-error-3-sxr-tune-failed-on-limesdr-usb/5303/16?u=dl9jbe).

In past, I was also asked to provide a **register dump**, which you can also find on the MyriadRF Forum [in a subsequent post of me](https://discourse.myriadrf.org/t/mcu-error-3-sxr-tune-failed-on-limesdr-usb/5303/27?u=dl9jbe).

cederom, 2021-01-23:
Aaahhh it was you @JanBeh asing the questions my bad please forgive me, multitasking kills focus! We need to make some radio transmissions to get familiar with our callsigns - it would be best to use LimeSDR and satellite or moon reflection :-) :-)

I noticed that hardware is very sensitive to a temperature changes and tests usually failed above 40C. Maybe you can try to see if the problem is anyhow related - it seems a bit cold outside so it could be a good time to test :-)

Otherwise I am not yet familiar enough with the solution to help you that deep sorry! The LimeSuite port is ready for FreeBSD.. maybe we should create a `limesuite-devel` to get stuff from git master if that helps anyhow? :-)

I just got the FreeBSD WIKI account and I have created a dedicated HAM Radio section :-)
https://wiki.freebsd.org/hamradio