myriadrf/LimeSuite #358: LimeUtil --update seg faults on Arch with libusb-1.0.25

horga83, 2022-03-08:
LimeUtil --update seg faults on Arch with libusb-1.0.25

If I downgrade libusb to 1.0.24 it runs fine.

arnobert, 2022-04-06:
Hi, I think this is the same error I get using SDRAngel (crashes when Lime connected during calling LMS_close() ).
73

horga83, 2022-04-06:
Yes it also happens with SDR++

arnobert, 2022-04-07:
And also with LimeSuite...

rjonaitis, 2022-04-07:
Crash is during closing connection by freeing never used transfer stuctures, really an issue of Libusb https://github.com/libusb/libusb/issues/1059#issuecomment-1040236517

jessupp, 2022-04-07:
I have also experience persistent segmentation fault on Raspberry PI Bullseye 32 and 64 bit latest OS.  Older Stretch and Buster OS  ok.  But this is concurring with Libusb1.0.24-3 and not version 1.0.25.  Below is what I get:

libusb-1.0-0-dev is already the newest version (2:1.0.24-3).

pi@raspberrypi:~/Desktop/ODR $ sudo odr-dabmod mod.ini
ODR-DabMod version v2.5.0, compiled at Apr  7 2022, 12:38:24
Compiled with features: zeromq output_uhd output_soapysdr 
Input
  Type: file
  Source: test1.eti
Output
 SoapySDR
  Device: LimeSDR-Mini
  master_clock_rate: 32768000
  Sampling rate: 2.0480 MHz
      Configuration parsed. Starting up version v2.5.0
      Soapy:Creating the device with: LimeSDR-Mini
[INFO] [UHD] linux; GNU C++ version 10.2.0; Boost_107100; UHD_3.15.0.0-4
Segmentation fault

jessupp, 2022-04-08:
Further to to above seg fault on raspberry pi, it it may not be libusb in my case:

Thread 1 "odr-dabmod" received signal SIGSEGV, Segmentation fault.
0x00074a98 in Output::Soapy::Soapy (this=0x10a01c, config=...)
    at src/output/Soapy.cpp:79
79	        m_device->getMasterClockRate()/1000.0 << " kHz";

closed 2022-04-08 by rjonaitis

rjonaitis, 2022-04-08:
Made a workaround for the 1.0.25 bug.

> I have also experience persistent segmentation fault on Raspberry PI Bullseye 32 and 64 bit latest OS. Older Stretch and Buster OS ok. But this is concurring with Libusb1.0.24-3 and not version 1.0.25.

Not sure what's happening there, as Libusb does not have official release of 1.0.24-3 https://github.com/libusb/libusb/tags , plus the 1.0.24 release did not have that bug.

jessupp, 2022-04-12:
> Made a workaround for the 1.0.25 bug.
>
>
> Not sure what's happening there, as Libusb does not have official release of 1.0.24-3 https://github.com/libusb/libusb/tags , plus the 1.0.24 release did not [...]

Thank you, you raise an interesting point with libusb releases - 1.0.24-3 is from Raspberry PI Bullseye repository.  There is a list at https://packages.qa.debian.org/libu/libusb-1.0.html - but neither do I understand these release numbers. 

However on a more positive note I pulled the latest changes of SoapySDR and Limesuite from Github (as I was using old 2019 versions).  Compiled these latest and all now works! No seg fault any more.  So for me looking like a Soapy/Limesuite old software not working on later Debian OS.  No problem with Libusb1.0.24-3 from Raspberry PI repository.

commit 9171098d (2022-04-08): Fix #358 ensure valid device handle for libusb_free_transfer to avoid crash if transfer was never used