myriadrf/LimeSuite #366: 22.09.0 build issue when Connection Remote is enabled.

cederom, 2022-09-08:
Hello world :-)

I am updating FreeBSD Port for LimeSuite after today's release 22.09.0. I have build error when  Remote Connection is enabled. Is it only me? :-)

```
/bin/mkdir -p /tmp/ports-build/usr/ports/comms/limesuite/work/.build
-- The C compiler identification is Clang 14.0.5
-- The CXX compiler identification is Clang 14.0.5
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at CMakeLists.txt:19 (cmake_policy):
  The OLD behavior for policy CMP0077 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Looking for pthread.h
-- Looking for pthread.h - found
-- Found Threads: TRUE
-- Performing Test HAVE_CXX_ATOMICS_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS_WITHOUT_LIB - Success
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB - Success
-- Found libusb-1.0:
--  - Includes: /usr/include
--  - Libraries: /usr/lib/libusb.so
-- USB_INCLUDE_DIRS: /usr/include
-- USB_LIBRARIES: /usr/lib/libusb.so
-- USB_INCLUDE_DIRS: /usr/include
-- USB_LIBRARIES: /usr/lib/libusb.so
-- Looking for wiringPi.h
-- Looking for wiringPi.h - not found
-- Found wxWidgets: -L/usr/local/lib;-pthread;;;-lwx_gtk3u_richtext-3.1;-lwx_baseu_xml-3.1;-lwx_gtk3u_html-3.1;-lwx_gtk3u_xrc-3.1;-lwx_gtk3u_aui-3.1;-lwx_gtk3u_core-3.1;-lwx_baseu-3.1;-lwx_gtk3u_gl-3.1 (found version "3.1.5")
-- wxWidgets_DEFINITIONS: _FILE_OFFSET_BITS=64;WXUSINGDLL;__WXGTK__;_THREAD_SAFE
-- wxWidgets_INCLUDE_DIRS: /usr/local/lib/wx/include/gtk3-unicode-3.1;/usr/local/include/wx-3.1
-- wxWidgets_LIBRARIES: -L/usr/local/lib;-pthread;;;-lwx_gtk3u_richtext-3.1;-lwx_baseu_xml-3.1;-lwx_gtk3u_html-3.1;-lwx_gtk3u_xrc-3.1;-lwx_gtk3u_aui-3.1;-lwx_gtk3u_core-3.1;-lwx_baseu-3.1;-lwx_gtk3u_gl-3.1
-- Found OpenGL: /usr/local/lib/libOpenGL.so
-- OPENGL_INCLUDE_DIR: /usr/local/include
-- OPENGL_LIBRARIES: /usr/local/lib/libOpenGL.so;/usr/local/lib/libGLX.so;/usr/local/lib/libGLU.so
-- Found Gnuplot: /usr/local/bin/gnuplot (found version "5.4.1")
-- GNUPlot Found
-- Found X11: /usr/local/include
-- Looking for XOpenDisplay in /usr/local/lib/libX11.so;/usr/local/lib/libXext.so
-- Looking for XOpenDisplay in /usr/local/lib/libX11.so;/usr/local/lib/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found FLTK: /usr/local/lib/libfltk_images.so;/usr/local/lib/libfltk_forms.so;/usr/local/lib/libfltk_gl.so;/usr/local/lib/libfltk.so
-- Module LMS7Support configured with version: 22.09.0
--
-- ######################################################
-- ## LimeSuite enabled features
[... 35 lines cut ...]
    CMAKE_C_FLAGS_DEBUG
    CMAKE_VERBOSE_MAKEFILE
    ENABLE_SIMD_FLAGS
    ENABLE_STREAM

-- Build files have been written to: /tmp/ports-build/usr/ports/comms/limesuite/work/.build
===>  Building for limesuite-22.09.0_4
(...)
[ 78% 123/156] : && /usr/bin/c++ -pipe -march=bdver2 -g -fstack-protector-strong -fno-strict-aliasing -pthread -pipe -march=bdver2 -g -fstack-protector-strong -fno-strict-aliasing -fstack-protector-strong src/CMakeFiles/LimeSuiteGUI.dir/ADF4002/ADF4002_wxgui.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlXBUF_view.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlRFE_view.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlCalibrations_view.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlTRF_view.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlCDS_view.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlLimeLightPAD_view.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlBIST_view.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/dlgViewIRAM.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_gui_utilities.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlSX_view.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlTBB_view.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_dlgVCOfrequencies.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlRBB_view.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlLDO_view.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlAFE_view.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlGains_view.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/CoefficientFileParser.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlRxTSP_view.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_mainPanel.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/dlgViewSFR.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlCLKGEN_view.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlMCU_BD_view.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlBIAS_view.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_dlgGFIR_Coefficients.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlTxTSP_view.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlR3.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/LMS_Programing/LMS_Programing_wxgui.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/utilities_gui/dlgConnectionSettings.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/utilities_gui/pnlMiniLog.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/utilities_gui/pnlAPI.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/utilities_gui/SPI_wxgui.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/utilities_gui/dlgFullMessageLog.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/utilities_gui/dlgDeviceInfo.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/utilities_gui/utilities_gui.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7suiteAppFrame.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/RFSpark_wxgui.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/Si5351C/Si5351C_wxgui.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7suiteApp_gui.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7suiteApp.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/FPGAcontrols_wxgui/FPGAcontrols_wxgui.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/dlgAbout.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/numericSlider/numericSlider.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/fftviewer_wxgui/fftviewer_wxgui.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/fftviewer_wxgui/fftviewer_frFFTviewer.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/lms7suiteEvents/lms7suiteEvents.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/pnlBoardControls.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/pnlQSpark.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/pnluLimeSDR.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/pnlUltimateEVB.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/pnlBuffers.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/pnlCoreSDR.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/pnlLimeNetMicro.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/__/external/kissFFT/kiss_fft.c.o src/CMakeFiles/LimeSuiteGUI.dir/windowFunction.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/pnlLimeSDR.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/pnlGPIO.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/limeRFE/limeRFE_gui.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/limeRFE/limeRFE_wxgui.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/limeRFE/limeRFE_cmd.cpp.o src/CMakeFiles/LimeSuiteGUI.dir/limeRFE/RFE_Device.cpp.o -o bin/LimeSuiteGUI -L/usr/local/lib -Wl,-rpath,/usr/local/lib:/tmp/ports-build/usr/ports/comms/limesuite/work/.build/src:  src/libLimeSuite.so.22.09.0  src/oglGraph/liboglGraph.a  -L/usr/local/lib  -pthread  -lwx_gtk3u_richtext-3.1  -lwx_baseu_xml-3.1  -lwx_gtk3u_html-3.1  -lwx_gtk3u_xrc-3.1  -lwx_gtk3u_aui-3.1  -lwx_gtk3u_core-3.1  -lwx_baseu-3.1  -lwx_gtk3u_gl-3.1  -lusb  -pthread  -lGL  /usr/local/lib/libOpenGL.so  /usr/local/lib/libGLX.so  /usr/local/lib/libGLU.so  -lwx_gtk3u_richtext-3. [... cut ...]
There seems to be some unresolved TODO at https://github.com/myriadrf/LimeSuite/blob/e829d3ede6222c1c0fc0e90dcd2050e50569e33d/src/ConnectionRemote/ConnectionRemoteEntry.cpp#L16?

```
octagon% cd limesuite/work/LimeSuite-22.09.0/
octagon% grep -r loadConnectionRemoteEntry *
src/ConnectionRemote/ConnectionRemoteEntry.cpp:void __loadConnectionRemoteEntry(void) //TODO fixme replace with LoadLibrary/dlopen
src/ConnectionRegistry/BuiltinConnections.in.cpp:void __loadConnectionRemoteEntry(void);
src/ConnectionRegistry/BuiltinConnections.in.cpp:    __loadConnectionRemoteEntry();
```

rjonaitis, 2022-09-09:
Remote connection is only for development/debugging purposes and is not efficient/safe or properly tested and can cause issues #263 , so it has been removed from the CMakeLists.txt to prevent it's inclusion to releases https://github.com/myriadrf/LimeSuite/blob/master/src/CMakeLists.txt#L162

The `-DENABLE_REMOTE` might be leftover in your cmake cache from previous builds, clear your CMake cache and that options should disappear and build will succeed.

closed 2022-09-09 by rjonaitis

cederom, 2022-09-09:
Thank you @rjonaitis for quick response and the new release! :-)

> The `-DENABLE_REMOTE` might be leftover in your cmake cache from previous builds, clear your CMake cache and that options should disappear and build will succeed.

The build is automated, everything is done from scratch, yes there was this option left from previous releases, now its clear this option is gone, one thing that still bothers me is the build error when this option is passed so it is still impacting something anywhow, can you please confirm? :-)

rjonaitis, 2022-09-09:
the only thing that define is impacting is the function call to load the remote connection: https://github.com/myriadrf/LimeSuite/blob/a45e482dad28508d8787e0fdc5168d45ac877ab5/src/ConnectionRegistry/BuiltinConnections.in.cpp#L45-L47

cederom, 2022-09-09:
Yup, then linker complains it cannot find the function.. up to you if that can be left alone.. just sharing my testing results :-)