myriadrf/LimeSuite #369: Issue with Debian packages.

9600, 2022-10-25:
# Summary

1. If I build and install Lime Suite from source using cmake + make, this works just fine.
2. If I build Lime Suite as a package, either using Debian tools locally or via Ubuntu PPA, I get errors when I run srsRAN eNodeB. 

The issue is that the libLimeSuite.so.22.09.1 which is built either via the MyriadRF PPA or locally with Debian packaging tools, results in _LMS_SetSampleRate: Failed to set RX sampling rate_ and _Too many trials_ being printed out. 

# Error details

andrew@lcran:~$ sudo srsenb /etc/srsran/enb.conf
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /etc/srsran/enb.conf...
WARNING: cpu0 scaling governor is not set to performance mode. Realtime processing could be compromised. Consider setting it to performance mode before running the application.

Built in RelWithDebInfo mode using 22.04.0.

/build/srsran-tSvzof/srsran-22.04LC01/srsenb/src/enb_cfg_parser.cc:1507: Force DL EARFCN for cell PCI=1 to 1934
connect(): Connection refused
Failed to initiate S1 connection. Attempting reconnection in 10 seconds
Opening 1 channels in RF device=lime with args=index=0,rxant=LNAH,txant=BAND2,cal=all
Supported RF device list: UHD lime file
Number of requested channels: 1
Found device #0: LimeSDR-USB, media=USB 3.0, module=FX3, addr=1d50:6108, serial=00090726074D041C
Reference clock 30.72 MHz
Initializing limesdr device
Setup RX stream 0
Setup TX stream 0
RX antenna/s set to: LNAH
TX antenna/s set to: BAND2

==== eNodeB started ===
Type <t> to view trace
LMS_SetSampleRate: Failed to set RX sampling rate
TX sampling rate: 3.84
Setting analog TX LPF BW to: 5.00
Setting manual TX/RX offset to 73 samples
Setting frequency: DL=1878.4 Mhz, UL=1783.4 MHz for cc_idx=0 nof_prb=15
Calibrating RX channel: 0, BW: 2.50
Calibrating TX channel: 0, BW: 2.50
Too many trials
Too many trials
^CStopping ..
Too many trials
Too many trials
Too many trials

# Additional notes

Configuring cmake with _-DCMAKE_BUILD_TYPE=RelWithDebInfo_ doesn't seem to trigger this problem, nor does stripping the library of symbols. Which were the only obvious differences I could see in how the library gets built as part of packaging.

ncorgan, 2023-06-13:
Are you sure the source-built versions match the packages? I built both package versions (LimeSuite tag v22.09.1, srsRAN release_22_04_LC01) from source, and I'm seeing the same behavior. This is without any of the custom flags `debhelper` adds when building the package.

Building off of master (a470dfecd4963c6edb1952c2d32a713650a5f463 as of this comment), I saw this with both source-built and package-generated versions.

9600, 2023-06-13:
Let me re-rest this week and get back to you.

ncorgan, 2023-07-07:
Were you able to verify the behavior I mentioned?

9600, 2023-07-28:
With srsRAN now building on Ubuntu 22.04 I was able to test with this built from source, and then Lime Suite packages installed from the PPA (22.09.1), and I don't see these errors. 

Not sure what the cause of the issue was when testing on Ubuntu 20.04 — and of course the srsRAN fork has now been updated to fix the build failure on Ubuntu 22.04 — but closing this issue.

closed 2023-07-28 by 9600