myriadrf/LimeSuite #371: Issue with rv64

FantasqueX, 2022-11-09:
Hi, I noticed that https://github.com/myriadrf/LimeSuite/pull/350 adds an exception for rv64 which solves the same issue with https://github.com/myriadrf/LimeSuite/pull/370. I should apologize here because I didn't notice that PR when I created my PR. I think one PR should be reverted, two snippets of code which try to solve the same problem is bad.

9600, 2022-11-09:
And a new issue raised? This is getting confusing :o) OK, hopefully @rjonaitis can resolve in the best way possible.

9600, 2022-11-09:
I meant opening this new issue, instead of commenting on #370.

FantasqueX, 2022-11-09:
It's a mistake actually. I should comment next time.

rjonaitis, 2022-11-15:
#370 can stay as a protective measure against -march=native usage where it is not supported, although I tried to check the `check_cxx_compiler_flag()` even with faulty parameters, it did not report any issues with them, so results might vary depeding on compiler.
#350 can also stay, but kind of needs modification, it currently just defines `set(DEFAULT_SIMD_FLAGS "rv64")`, but this flag is not matched in [CMakeLists.txt#L139-L157](https://github.com/FantasqueX/LimeSuite/blob/436b7a7f71097e403d883cab123435941c8eeba3/CMakeLists.txt#L139-L157) so essentially it just avoids the '-march=native' case, but does not actually add definition to compile with -march=rv64

I don't have RISC-V board or cross compile tools setup right now, so can't actually verify what the compiler is doing then.

FantasqueX, 2022-11-16:
> #370 can stay as a protective measure against -march=native usage where it is not supported, although I tried to check the `check_cxx_compiler_flag()` even with faulty parameters, it did not report any issues with them, so results might vary depeding on compiler.

What faulty parameters did you try? I think it may relate to an issue with CMake. However, it works well `-march=native`.

> #350 can also stay, but kind of needs modification, it currently just defines `set(DEFAULT_SIMD_FLAGS "rv64")`, but this flag is not matched in [...]

I think those changes can be reverted. We (ArchLinux RISC-V) add compile flags `-march=rv64gc` when packaging. And gcc has a default march value when there is no explicit march argument. I'm quite sure LimeSuite can be built well when there is no match case.