myriadrf/LimeSuite #378: Firmware missing upstream

Dapid, 2023-06-05:
`LimeUtil --update` fails because the firmware is missing from upstream.

```
Connected to [LimeSDR Mini [USB 3.0] 1D90F927C75C9D]
--2023-06-05 11:25:24--  https://downloads.myriadrf.org/project/limesuite/22.09/LimeSDR-Mini_HW_2.0_r2.2.bit
Resolving downloads.myriadrf.org (downloads.myriadrf.org)... 165.227.233.124, 2a03:b0c0:1:d0::eed:8001
Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|165.227.233.124|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-06-05 11:25:24 ERROR 404: Not Found.

Failed: wget --output-document="/home/david/.local/share/LimeSuite/images/22.09/LimeSDR-Mini_HW_2.0_r2.2.bit" "https://downloads.myriadrf.org/project/limesuite/22.09/LimeSDR-Mini_HW_2.0_r2.2.bit"

Programming update failed!
```

A 2.0 is available in the folder corresponding to [20.10][1].

   [1]: https://downloads.myriadrf.org/project/limesuite/20.10/

9600, 2023-06-05:
LimeSDR Mini 2.0 is not supported by Lime Suite release 22.09. A new release is in the pipeline, but for now you should build from git.

Dapid, 2023-06-05:
Thanks for the quick reply.

My Lime has v 2.0, am I correct in understanding that may cause incompatibility problems? Would you recommend downgrading LimeSuite to 20.10, or building from master?

9600, 2023-06-05:
Build from git master.

jurajbujdoso, 2023-07-28:
Will do for sure. But this make me really sad.
Since I have wait for v2 almost a year. Version 1 has been canceled and version 2 deliver with several excuses in delivery.

We have been tolerant and does not blame company. In general chip shortage etc.
One of decision to buy Lime was, that I have not think about the support for sw in FMO.

The only limit which I can see now,  is currently on your side and does not have any relation to 3rd vendor.

And I can not really understand, why there is no support for "latest product" as it was in the past.

Or does the every product you deliver sucks to get updates? 
 provides error messages without excuses and really DO IT YOURSELF work.

9600, 2023-11-28:
23.11 release includes support and images are in the download server.

closed 2023-11-28 by 9600