myriadrf/LimeSuite #413: Cannot configure Limesuite 23.11 on Ubuntu Jammy 22.04.5 LTS

sayboltm, 2025-03-06:
I cannot install OR remove limesuite, or any package that interfaces with any of its dependencies.  As soon as it gets to here: 

I get:
```

--2025-03-06 16:21:33--  (try:16)  http://downloads.myriadrf.org/project/limesuite/23.11/
Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|2a03:b0c0:1:d0::eed:8001|:80... connected.
HTTP request sent, awaiting response... Read error (Connection reset by peer) in headers.
Retrying.

--2025-03-06 16:21:43--  (try:17)  http://downloads.myriadrf.org/project/limesuite/23.11/
Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|2a03:b0c0:1:d0::eed:8001|:80... connected.
HTTP request sent, awaiting response... Read error (Connection reset by peer) in headers.
Retrying.

--2025-03-06 16:21:53--  (try:18)  http://downloads.myriadrf.org/project/limesuite/23.11/
Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|2a03:b0c0:1:d0::eed:8001|:80... connected.
HTTP request sent, awaiting response... Read error (Connection reset by peer) in headers.
Retrying.
```
...
```
dpkg: error processing package limesuite-images23.11:amd64 (--configure):
 installed limesuite-images23.11:amd64 package post-installation script subprocess returned error exit status 4
dpkg: dependency problems prevent configuration of limesuite-images:amd64:
 limesuite-images:amd64 depends on limesuite-images23.11; however:
  Package limesuite-images23.11:amd64 is not configured yet.

dpkg: error processing package limesuite-images:amd64 (--configure):
 dependency problems - leaving unconfigured
No apport report written because the error message indicates its a followup error from a previous failure.
                                                                                                          Processing triggers for man-db (2.10.2-1) ...
Errors were encountered while processing:
 limesuite-images23.11:amd64
 limesuite-images:amd64
E: Sub-process /usr/bin/dpkg returned an error code (1)
W: Operation was interrupted before it could finish

```

This seems to be related to the insecure connection, I reproduced using wget:
Steps to reproduce:
```
wget http://downloads.myriadrf.org/project/limesuite/23.11/
--2025-03-06 16:24:20--  http://downloads.myriadrf.org/project/limesuite/23.11/
Resolving downloads.myriadrf.org (downloads.myriadrf.org)... 2a03:b0c0:1:d0::eed:8001, 165.227.233.124
Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|2a03:b0c0:1:d0::eed:8001|:80... connected.
HTTP request sent, awaiting response... Read error (Connection reset by peer) in headers.
Retrying.
```

I cannot even remove the package:

```
Removing soapysdr0.8-module-rfspace:amd64 (0.2.5-6) ...
Removing soapysdr0.8-module-rtlsdr:amd64 (0.3.2-1) ...
Removing soapysdr0.8-module-uhd:amd64 (0.4.1-3build1) ...
Setting up limesuite-images23.11:amd64 (23.11.0-1) ...
Downloading http://downloads.myriadrf.org/project/limesuite/23.11/ -> /usr/share/LimeSuite/images/23.11
--2025-03-06 16:19:44--  http://downloads.myriadrf.org/project/limesuite/23.11/
Resolving downloads.myriadrf.org (downloads.myriadrf.org)... 2a03:b0c0:1:d0::eed:8001, 165.227.233.124
Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|2a03:b0c0:1:d0::eed:8001|:80... connected.
HTTP request sent, awaiting response... Read error (Connection reset by peer) in headers.
Retrying.

```

Is there an issue with the website or a possible workaround to point it somewhere else without re-configuring this machine to build from source? 

Thanks

sayboltm, 2025-03-06:
It does appear the website is not working in a browser either, and I cannot get the deps to build from source per here:

https://wiki.myriadrf.org/Lime_Suite#Building_from_source

argh! Doesn't look too bad if I already had them downloaded.

rjonaitis, 2025-03-06:
tested just now, the website is working http://downloads.myriadrf.org
Might have been a temporary outage.

sayboltm, 2025-03-06:
Really?
I did too, it does not seem to be working:

[image]

rjonaitis, 2025-03-06:
Does it work for you if you specify the URL to be https?

sayboltm, 2025-03-06:
nope, won't work with wget that way either.

rjonaitis, 2025-03-06:
Does not look like a server side issue. I'm being properly redirected to https, browsing and files download using wget works as expected.
Do you have enabled firewall?

rjonaitis, 2025-03-06:
Your connection seems to be using ipv6 address:
`Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|2a03:b0c0:1:d0::eed:8001|:80... connected.`

I'm connecting using ipv4
`Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|165.227.233.124|:443... connected.`
Unfortunately my internet provider does not support ipv6, so I can't verify if ipv6 is the issue.

@9600 Could you check if the site is fully configured to be accessible using ipv6?

sayboltm, 2025-03-07:
I see a possible typo? maybe this is just a status indicator but:

> Your connection seems to be using ipv6 address: `Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|2a03:b0c0:1:d0::eed:8001|:80... connected.`
> Your connection seems to be using ipv6 address: `Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|2a03:b0c0:1:d0::**eed:8001|:80...** connected.`

This looks like duplicate ports or something/leftover, idk. Vs:

> I'm connecting using ipv4 `Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|165.227.233.124|:443... connected.` Unfortunately my internet provider does not support ipv6, so I can't verify if ipv6 is the issue.
> I'm connecting using ipv4 `Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|165.227.233.**124|:443...** connected.` Unfortunately my internet provider does not support ipv6, so I can't verify if ipv6 is the issue.

This looks normal, or more normal.

> 
> [@9600](https://github.com/9600) Could you check if the site is fully configured to be accessible using ipv6?

Sure enough, forcing ipv4 fixes in wget:

`wget --inet4-only wget https://downloads.myriadrf.org/project/limesuite/23.11/
--2025-03-06 19:05:53--  http://wget/
Resolving wget (wget)... failed: Name or service not known.
wget: unable to resolve host address ‘wget’
--2025-03-06 19:05:53--  https://downloads.myriadrf.org/project/limesuite/23.11/
Resolving downloads.myriadrf.org (downloads.myriadrf.org)... 165.227.233.124
Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|165.227.233.124|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: ‘index.html’

index.html              [ <=>                ]   4.83K  --.-KB/s    in 0.03s   

2025-03-06 19:05:54 (162 KB/s) - ‘index.html’ saved [4948]

FINISHED --2025-03-06 19:05:54--
Total wall clock time: 0.8s
Downloaded: 1 files, 4.8K in 0.03s (162 KB/s)
`

I would bet if I find a non-intrusive way to force dpkg it may fix it. I will report back. 

Thank you **_very much_** for your help and rapid response!

So the issue is either with the site/server config RE IPv6 @9600  or within LimeSuite or within dpkg's handling of IPV6?

sayboltm, 2025-03-07:
I do think it is still IPv6 related but, unfortunately there are difficulties forcing IPv4.  The stdout during the query to the LimeSuite PPA looks identical to the wget tests, do you know is it actually calling wget? wget alone is forced to IPv4 and resolved, no problem (`technicalDebt++;`). 

To force `dpkg` requires:
`sudo apt-get -o Acquire::ForceIPv4=true update`

I skipped to trying to install my now-broken package that has nothing to do with LimeSuite: 
`sudo apt-get -o Acquire::ForceIPv4=true install apt-offline --no-install-recommends`
But this did not fix the issue:
```
sudo apt-get -o Acquire::ForceIPv4=true install apt-offline --no-install-recommends
[sudo] password for user: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
apt-offline is already the newest version (1.8.4-1).
The following packages were automatically installed and are no longer required:
  liblimesuite20.10-1 liblimesuite23.11-1 libmirisdr0 libosmosdr0
  limesuite-images limesuite-images23.11 limesuite-udev
  python3-proton-keyring-linux-secretservice
  python3-proton-vpn-killswitch-network-manager
  python3-proton-vpn-killswitch-network-manager-wireguard
  python3-proton-vpn-logger soapyosmo-common0.8
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up limesuite-images23.11:amd64 (23.11.0-1) ...
Downloading http://downloads.myriadrf.org/project/limesuite/23.11/ -> /usr/share/LimeSuite/images/23.11
--2025-03-06 19:21:37--  http://downloads.myriadrf.org/project/limesuite/23.11/
Resolving downloads.myriadrf.org (downloads.myriadrf.org)... 2a03:b0c0:1:d0::eed:8001, 165.227.233.124
Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|2a03:b0c0:1:d0::eed:8001|:80... connected.
HTTP request sent, awaiting response... Read error (Connection reset by peer) in headers.
Retrying.

--2025-03-06 19:21:38--  (try: 2)  http://downloads.myriadrf.org/project/limesuite/23.11/
Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|2a03:b0c0:1:d0::eed:8001|:80... connected.
HTTP request sent, awaiting response... Read error (Connection reset by peer) in headers.
Retrying.
```
... 5 minutes later:
```
--2025-03-06 19:24:06--  (try:20)  http://downloads.myriadrf.org/project/limesuite/23.11/
Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|2a03:b0c0:1:d0::eed:8001|:80... connected.
HTTP request sent, awaiting response... Read error (Connection reset by peer) in headers.
Giving up.

dpkg: error processing package limesuite-images23.11:amd64 (--configure):
 installed limesuite-images23.11:amd64 package post-installation script subprocess returned error exit status 4
dpkg: dependency problems prevent configuration of limesuite-images:amd64:
 limesuite-images:amd64 depends on limesuite-images23.11; however:
  Package limesuite-images23.11:amd64 is not configured yet.

dpkg: error processing package limesuite-images:amd64 (--configure):
 dependency problems - leaving unconfigured
No apport report written because the error message indicates its a followup error from a previous failure.
                          Errors were encountered while processing:
 limesuite-images23.11:amd64
 limesuite-images:amd64
E: Sub-process /usr/bin/dpkg returned an error code (1)
W: Operation was interrupted before it could finish
```

So I tried the persistent apt option: shown [here](https://docs.vultr.com/force-apt-get-to-ipv4-or-ipv6-on-ubuntu-or-debian)
create file: /etc/apt/apt.conf.d/99force-ipv4
containing:
`Acquire::ForceIPv4 "true";`
alternatively containing: (though my `vim` does not like the syntax)
`Acquire::ForceIPv4=true;`
or in a faster w1:r0 (write once, read zero) style syntax:
[... 33 lines cut ...]
<ins>

prints the following:
```
	Hit:5 https://dl.google.com/linux/chrome/deb stable InRelease                  
	Hit:6 http://us.archive.ubuntu.com/ubuntu jammy-backports InRelease            
	Ign:7 https://www.kismetwireless.net/repos/apt/release/cosmic cosmic InRelease 
	Err:8 https://www.kismetwireless.net/repos/apt/release/cosmic cosmic Release
	  404  Not Found [IP: 165.227.176.149 443]
	Hit:9 https://repo.protonvpn.com/debian stable InRelease            
	Ign:10 https://ppa.launchpadcontent.net/bladerf/bladerf/ubuntu jammy InRelease
	Ign:11 https://ppa.launchpadcontent.net/gqrx/gqrx-sdr/ubuntu jammy InRelease
	Hit:12 https://ppa.launchpadcontent.net/myriadrf/drivers/ubuntu jammy InRelease
	Ign:13 https://ppa.launchpadcontent.net/myriadrf/gnuradio/ubuntu jammy InRelease
	Err:14 https://ppa.launchpadcontent.net/bladerf/bladerf/ubuntu jammy Release
	  404  Not Found [IP: 185.125.190.80 443]
	Err:15 https://ppa.launchpadcontent.net/gqrx/gqrx-sdr/ubuntu jammy Release
	  404  Not Found [IP: 185.125.190.80 443]
	Err:16 https://ppa.launchpadcontent.net/myriadrf/gnuradio/ubuntu jammy Release
	  404  Not Found [IP: 185.125.190.80 443]
	Reading package lists... Done
```
Some of the myriadrf hits are being resolved to Canonical's datacentre in the UK, but some 404. What's up with that, eh?

In any event, this shows that there is an issue with IPv6, I don't have a way to reliably force IPv4 (yet), it is still trying to use IPv6 to access myriadrf.org code, which seems to be causing an issue.

sayboltm, 2025-03-07:
Some people say it doesn't work, but I tried editing `/etc/gai.conf` per [this post explaining](https://unix.stackexchange.com/questions/9940/convince-apt-get-not-to-use-ipv6-method), which "works at a much lower level (DNS and IP networking) than APT, so it will change how all your applications network--at least, all that use `getaddrinfo`"

(they mean `ping`, `host`, etc.)

Before:
``
ping halifax.myriadrf.org
PING halifax.myriadrf.org(halifax.myriadrf.org (2a03:b0c0:1:d0::eed:8001)) 56 data bytes
64 bytes from halifax.myriadrf.org (2a03:b0c0:1:d0::eed:8001): icmp_seq=1 ttl=48 time=176 ms
64 bytes from halifax.myriadrf.org (2a03:b0c0:1:d0::eed:8001): icmp_seq=2 ttl=48 time=159 ms
``
After adding: `precedence 2001:470::/96 100`:
to: `/etc/gai.conf`
in order to: Case 3: prefer IPV4 for specific hosts
and trying the same thing again despite it not working the first time after the edit:
``
124) 56(84) bytes of data.
64 bytes from halifax.myriadrf.org (165.227.233.124): icmp_seq=1 ttl=48 time=157 ms
64 bytes from halifax.myriadrf.org (165.227.233.124): icmp_seq=2 ttl=48 time=349 ms
``
It "updates" and takes the change. Plausibly working!

Alas, the issue persists. does `dpkg` or LimeSuite use `getaddrinfo` (or `wget`)?

sayboltm, 2025-03-07:
It seems `myriadrf.org` does have an IPv4 address I can reach in the USA, and I don't completely understand the `/etc/gai.conf`

Removing my change: It is back to IPv6
``
ping halifax.myriadrf.org
PING halifax.myriadrf.org(halifax.myriadrf.org (2a03:b0c0:1:d0::eed:8001)) 56 data bytes
64 bytes from halifax.myriadrf.org (2a03:b0c0:1:d0::eed:8001): icmp_seq=1 ttl=48 time=161 ms
``

Cranking it up: Case 1: prefer IPV4 for all hosts (I am still the same host? SO:/u/[Lmwangi](https://unix.stackexchange.com/users/7525/lmwangi) words not mine)
adding: `precedence ::ffff:0:0/96  100`
instead of previous: `precedence 2001:470::/96 100` 
to my
`/etc/gai.conf`
results in instant change of plausibly behavior (no refresh/source activate/ip up/down etc. required):
``
ping halifax.myriadrf.org
PING halifax.myriadrf.org (165.227.233.124) 56(84) bytes of data.
64 bytes from halifax.myriadrf.org (165.227.233.124): icmp_seq=1 ttl=48 time=152 ms
``
and THAT DID IT: 
```
apt-offline is already the newest version (1.8.4-1).
The following packages were automatically installed and are no longer required:
  liblimesuite20.10-1 liblimesuite23.11-1 libmirisdr0 libosmosdr0
  limesuite-images limesuite-images23.11 limesuite-udev
  python3-proton-keyring-linux-secretservice
  python3-proton-vpn-killswitch-network-manager
  python3-proton-vpn-killswitch-network-manager-wireguard
  python3-proton-vpn-logger soapyosmo-common0.8
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up limesuite-images23.11:amd64 (23.11.0-1) ...
Downloading http://downloads.myriadrf.org/project/limesuite/23.11/ -> /usr/share
/LimeSuite/images/23.11
--2025-03-06 21:32:38--  http://downloads.myriadrf.org/project/limesuite/23.11/
Resolving downloads.myriadrf.org (downloads.myriadrf.org)... 165.227.233.124, 2a
03:b0c0:1:d0::eed:8001
Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|165.227.233.124|:8
0... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://downloads.myriadrf.org/project/limesuite/23.11/ [following]
--2025-03-06 21:32:39--  https://downloads.myriadrf.org/project/limesuite/23.11/
Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|165.227.233.124|:4
43... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: ‘/tmp/tmp.pMQNk6NYRA/index.html.tmp’

index.html.tmp          [ <=>                ]   4.83K  --.-KB/s    in 0.02s   

2025-03-06 21:32:39 (240 KB/s) - ‘/tmp/tmp.pMQNk6NYRA/index.html.tmp’ saved [494
8]

Loading robots.txt; please ignore errors.
--2025-03-06 21:32:39--  https://downloads.myriadrf.org/robots.txt
Reusing existing connection to downloads.myriadrf.org:443.
HTTP request sent, awaiting response... 404 Not Found
2025-03-06 21:32:39 ERROR 404: Not Found.

Removing /tmp/tmp.pMQNk6NYRA/index.html.tmp since it should be rejected.

--2025-03-06 21:32:39--  https://downloads.myriadrf.org/project/limesuite/23.11/
LimeNET-Micro_lms7_trx_HW_2.1_r1.3.rpd
Reusing existing connection to downloads.myriadrf.org:443.
HTTP request sent, awaiting response... 200 OK
Length: 577536 (564K) [application/octet-stream]
[... 110 lines cut ...]
LimeSDR-USB_HW_1.4_ 100%[===================>] 565.73K  --.-KB/s    in 0.06s   

2025-03-06 21:32:42 (8.74 MB/s) - ‘/tmp/tmp.pMQNk6NYRA/LimeSDR-USB_HW_1.4_r2.23.
rbf’ saved [579307/579307]

--2025-03-06 21:32:42--  https://downloads.myriadrf.org/project/limesuite/23.11/
LimeSDR-USB_HW_1.4_r4.0.img
Reusing existing connection to downloads.myriadrf.org:443.
HTTP request sent, awaiting response... 200 OK
Length: 185976 (182K) [application/octet-stream]
Saving to: ‘/tmp/tmp.pMQNk6NYRA/LimeSDR-USB_HW_1.4_r4.0.img’

LimeSDR-USB_HW_1.4_ 100%[===================>] 181.62K  --.-KB/s    in 0.05s   

2025-03-06 21:32:43 (3.53 MB/s) - ‘/tmp/tmp.pMQNk6NYRA/LimeSDR-USB_HW_1.4_r4.0.i
mg’ saved [185976/185976]

FINISHED --2025-03-06 21:32:43--
Total wall clock time: 4.6s
Downloaded: 12 files, 5.5M in 1.3s (4.17 MB/s)
Moving images to /usr/share/LimeSuite/images/23.11
Setting up limesuite-images:amd64 (23.11.0-1) ...
```

It looks like `limesuite` has got what it needed to finish setting up and not interfere with other packages... and now I can probably use LimeSuite to better my alternate-frequency wireless skills instead of doing what I'm supposed to be doing! Cool!