myriadrf/LimeSuite #415: Ship the udev rules upstream

AsciiWolf, 2025-04-21:
Please, consider submitting the [udev rules](https://github.com/myriadrf/LimeSuite/blob/master/udev-rules/64-limesuite.rules) to upstream (systemd). Here is an example how to do it: https://github.com/systemd/systemd/pull/20561

Without this, it is [not possible](https://github.com/flatpak/xdg-desktop-portal/issues/227#issuecomment-946760482) to use LimeSDR devices in Flatpak and other special configurations where it is not possible to install the LimeSuite libs/udev rules system-wide.

Please see the upstream ticket: https://github.com/systemd/systemd/issues/37019

rjonaitis, 2025-04-22:
The rules contain VendorIDs/ProductIDs that belong to other vendors/devices, so it wouldn't be good if they become part of systemd and have side effect of getting applied to other devices that are not LimeSDR.

AsciiWolf, 2025-04-22:
>it wouldn't be good if they become part of systemd

Why? It would just make other devices work out-of-box.

rjonaitis, 2025-04-22:
It sounds convenient, but there are security reasons why devices have limited permissions by default.
If the rules would be affecting just LimeSDR devices, then yes, by all means, systemd update would be encouraged.
But because these rules could potentially change permissions of other devices for all other users, that have nothing to do with LimeSDR, then it's a problem.

AsciiWolf, 2025-05-13:
If you ever re-consider this, bladeRF was already added into systemd hwdb and can be used as an example of how the changes should look like:

https://github.com/systemd/systemd/blob/main/hwdb.d/70-software-radio.hwdb